Klipper3d/klipper #1018: What is Development Environment?

rmcbc, 2018-12-19:
@KevinOConnor can you please tell what is your development environment?

- For python I use PyCharm, what do you use that fits perfect in klipper code?
- For c/c++ normaly I use arduino but because this c/c++ is for diferent processors what IDE do you use?

mcmatrix, 2018-12-20:
I'm using https://code.visualstudio.com/

rmcbc, 2018-12-20:
@mcmatrix this is a good IDE, on windows to compile what you use for C/C++? Or simply use the raspberry compiler?

mcmatrix, 2018-12-21:
@rmcbc Yep, simply using the raspberry compiler.

KevinOConnor, 2018-12-21:
FWIW, I don't use an IDE.  I just use a regular editor (emacs).

-Kevin

KevinOConnor, 2018-12-27:
I'm going to close this as I think the original question was answered.

-Kevin

closed 2018-12-27 by KevinOConnor