Klipper3d/klipper #1019: Printing pauses for no obvious reason

muenznej, 2018-12-19:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2696470/klippy.log)

Hi, i have a strange problem with printing and i think its klipper/firmware related.

I wanted to test the" minimum print time" per layer and the highest acceptble "minimum speed" per layer with a small tube model. It has 5 mm diamter and is tested for various speeds.

It all works fine up to about 20 mm/s @ .2mm dz and then it starts to stop the printing on more or less regular time intervals ( makign ugly blobs ). I suspect the AVR(ATMEGA1284,  ender3) to be too busy to process the commands but i dont know how to tell.

As a alteration i  added retract at every layer change and that imediatly solved the issue for up to 50mm/s which is my current flowrate limit.

I will test a small square tower to test wheater the command list is too exessive with theese curves. 

I hope you could shine some light on this wierd issue!

thanks for all the work. go klipper go klipper go :dancing_men:

jakep82, 2018-12-19:
Have you tried virtual SD?  Octoprint has well known limitations on analysis/transfer speeds which become evident when you print round objects with short line segments.

muenznej, 2018-12-19:
Hi, i just now tested the rectangular tower with same lengths and it does NOT show this problem. So i guess it has to do with the curves.

> Have you tried virtual SD? Octoprint has well known limitations on analysis/transfer speeds which become evident when you print round objects with short line segments.

i am not quite sure what you mean. I am using octoprint and i use cura to send it to octoprint. can you explain to me how i should proceed to test this "virtual sd"?

BlackStump, 2018-12-19:
Octoprint is the bottleneck the virtual SD bypasses Octoprint
Octoprint is I believe trying to fix the comms issue but not until they release 1.4

muenznej, 2018-12-19:
> Octoprint is the bottleneck the virtual SD bypasses Octoprint
> Octoprint is I believe trying to fix the comms issue but not until they release 1.4
> 
> if you are using a low powered host you will suffer more ie raspi 1 /zero will struggle with Octoprint wanting more cpu cycles then is available

ok thanks!

I thought raspi 3b+ was good enough :/ Well that workaround will help for sure especially when i have organic high detail models! thanks

closed 2018-12-19 by muenznej