Klipper3d/klipper #1020: lost steps detection & correction for TMC 2208 driver?

megablue, 2018-12-20:
In the Endstop_Phase.md doc,  it mentioned you can also use endstop phase on non-tranamic drivers, which got me into thinking is it possible combine it with the Index Output on tmc2208 to server as a rudimentary "closed loop" lost steps detection? since you can count steps manually and compare it with the index provided by the tmc drivers.

yschroeder, 2018-12-20:
AFAIK the index signal of the stepper driver does not take lost steps into account. It solely indicates that the driver went through the full microstep table and is back at the beginning. The index signal will also trigger regularly if you clamp down the motor shaft.

KevinOConnor, 2018-12-27:
I'm going to close this as I think the original question was answered (alas, no, if the tmc2208 loses steps it doesn't know it, so no way to get the info from it).

-Kevin

closed 2018-12-27 by KevinOConnor