Klipper3d/klipper #1021: Not work on Arduino DUE

semos2k, 2018-12-20:
I have tried the following steps:

`$ make
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/sam3x8e/main.o
  Compiling out/src/sam3x8e/timer.o
  Compiling out/src/sam3x8e/gpio.o
  Compiling out/src/sam3x8e/spi.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/alloc.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/timer_irq.o
  Compiling out/src/../lib/sam3x/gcc/system_sam3xa.o
  Compiling out/src/../lib/sam3x/gcc/gcc/startup_sam3xa.o
  Compiling out/src/sam3x8e/serial.o
  Compiling out/src/generic/serial_irq.o
  Building out/compile_time_request.o
Version: v0.6.0-674-gd0f0972-20181219_225507-semos2k-mcu
  Linking out/klipper.elf
  Creating hex file out/klipper.bin
  
$ make flash FLASH_DEVICE=/dev/ttyACM0
  Flashing out/klipper.bin lib/bossac/bin/bossac to /dev/ttyACM0 via bossac
Erase flash

Done in 0.075 seconds
Write 15824 bytes to flash (62 pages)
[==============================] 100% (62/62 pages)
Done in 3.878 seconds
Verify 15824 bytes of flash
[==============================] 100% (62/62 pages)
Verify successful
Done in 3.559 seconds
Set boot flash true`

and when it is started klipper shows:

`[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100
=======================
Extruder max_extrude_ratio=0.266081
Starting serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect`

semos2k, 2018-12-20:
This is my actual config and i used the programing port:

[mcu]
/dev/serial/by-path/pci-0000:00:06.0-usb-0:2:1.0
pin_map: arduino

[stepper_x]
step_pin: ar24
dir_pin: ar23
enable_pin: ar26
step_distance: .0125
endstop_pin: ^ar28
#endstop_pin: ^ar34
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_y]
step_pin: ar17
dir_pin: !ar16
enable_pin: ar22
step_distance: .0125
endstop_pin: ^ar30
#endstop_pin: ^ar36
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_z]
step_pin: ar2
dir_pin: ar3
enable_pin: ar15
step_distance: .0025
endstop_pin: ^ar32
#endstop_pin: ^ar38
position_endstop: 0.5
position_max: 200

[extruder]
step_pin: analog7
dir_pin: analog6
enable_pin: analog8
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar13
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog0
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250

#[extruder1]
#step_pin:           analog10
#dir_pin:            analog9
#enable_pin:         analog11

#[extruder2]
#step_pin:           ar51
#dir_pin:            ar53
#enable_pin:         ar49

[heater_bed]
heater_pin: ar7
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog1
control: watermark
min_temp: 0
max_temp: 130

[fan]
pin: ar9

#[heater_fan nozzle_cooling_fan]
#pin: ar8

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

jakep82, 2018-12-20:
If you're using a clone Due, you will need to either modify the Arduino by adding a resistor, flashing the 16u2 USB serial converter with modified firmware, or press the reset button by hand.  Pretty much every clone has this problem.  See the linked thread for more information.  You can also buy a genuine Due which doesn't have this problem.

https://forum.arduino.cc/index.php?topic=256771.0

yschroeder, 2018-12-20:
I started writing something while @jakep82 came up with a hopefully helpful solution. But I will post my findings nonetheless so it might be helpful for someone else finding this...

The programming port should be the correct one.

Here are some more details I figured out:
According to the [schematics](https://www.arduino.cc/en/uploads/Main/arduino-Due-schematic.pdf) the ATmega16U2 that is used as a USB/UART bridge is connected to the pins `PA8` and `PA9` which is the same UART that Klipper uses (see [here](https://github.com/KevinOConnor/klipper/blob/master/src/sam3x8e/serial.c#L16)). The ATmega16U2 sets the baudrate of its hardware UART to the value configured via USB (see [here](https://github.com/arduino/ArduinoCore-sam/blob/master/firmwares/atmega16u2/arduino-usbserial/Arduino-usbserial.c#L273)). So it should work like any other USB/UART bridge (e.g. FTDI).

I found nothing wrong, I just wanted to have this documented in case somebody else runs into this again.

semos2k, 2018-12-20:
I use a clone, I tried pressing the reset button but the result is the same.

I'm going to test the direct connection to the pins to see if the firmware is being executed.

jakep82, 2018-12-20:
If I remember correctly you need to press the reset button, then issue a firmware_restart right after.  I have a clone and found it difficult to get it connected unless I timed it right.  I eventually got a genuine Due because pressing the reset button was annoying.

semos2k, 2018-12-21:
I think it may be my klipper binary file is corrupted or it is not compiled correctly.

ktotomskru, 2018-12-22:
I ran into absolutely the same issue. Everything worked fine until about a week ago. 
I have two arduino due boards (clones) with ruramps4d shield. I had no issue with flashing them with make flash blah blah blah, but yesterday i tried to reflash them with fresh klipper version with the same sad result:

> Starting serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

I tried flashing from Orange PI PC with armbian (4.14.78) and from Rock64 with ubuntu 18.04 (4.4.132-1072-rockchip-ayufan-ga1d27dba5a2e). Result is absolutely the same :(

ktotomskru, 2018-12-22:
tried flashing manually - no luck:

> root@orangepipc:/home/printer/klipper/out#

 bossac -e -R -w -v -i -b ./klipper.bin --port=ttyACM0 
Erase flash
Write 15820 bytes to flash
[==============================] 100% (62/62 pages)
Verify 15820 bytes of flash
[==============================] 100% (62/62 pages)
Verify successful
Set boot flash true
Device       : ATSAM3X8
Chip ID      : 285e0a60
Version      : v1.1 Dec 15 2010 19:25:04
Address      : 524288
Pages        : 2048
Page Size    : 256 bytes
Total Size   : 512KB
Planes       : 2
Lock Regions : 32
Locked       : none
Security     : false
Boot Flash   : true
CPU reset.

ktotomskru, 2018-12-22:
Just in case. I'm aware of issue that reqiures non-original due to be reset after flashing. I tried.

semos2k, 2018-12-22:
I have used two versions of Bossac which is installed with klipper(1.9) and the one that comes with Arduino IDE(1.6.1-arduino), both have generated the same behavior.

semos2k, 2018-12-22:
Please allow me access to the file (semos2k@gmail.com). 
thx.

ktotomskru, 2018-12-22:
> Please allow me access to the file ([semos2k@gmail.com](mailto:semos2k@gmail.com)).
> thx.

please try again, i've corrected permissions

ktotomskru, 2018-12-22:
Hi

please try again, it must be available now

semos2k, 2018-12-22:
Yes, thx.

semos2k, 2018-12-22:
If this firmware works correctly ..... thx, in my tests I use an x86 linux.
Did you use any special parameter? @ktotomskru

ktotomskru, 2018-12-22:
> If this firmware works correctly ..... thx, in my tests I use an x86 linux.
> Did you use any special parameter? @ktotomskru

nothing special, just git clone and make. host is ubuntu 16.04 (4.13.0-37-generic).

KevinOConnor, 2018-12-22:
> I tried flashing from Orange PI PC with armbian (4.14.78) and from Rock64 with ubuntu 18.04 

FYI, The arm compiler on Ubuntu 18.04 is known to be broken - see #443.

-Kevin

semos2k, 2018-12-23:
if you're right, I've verified it and it's because of the linux version, I found a temporary solution in stackoverflow.

https://unix.stackexchange.com/questions/453032/how-to-install-a-functional-arm-cross-gcc-toolchain-on-ubuntu-18-04

thx, @KevinOConnor @ktotomskru

closed 2018-12-23 by semos2k