Klipper3d/klipper #1026: mesh leveling problems after klipper upgrade

ifUpEth0, 2018-12-21:
I recently did a pull/upgrade of my klipper install. I am having odd issues now with Mesh leveling. It is easiest to see with several pieces being printed front to back of the machine. The skirting and first layer perimeter go down well. Most of the infill will go down well but will be just a touch too high (z) toward the front of the printer. The really strange part is that I will see this in each piece progressively each layer. For example 4 pieces in a straight line down the middle of the printer will all have thinned or raised front left infill  and (perimeters after the first layer). Given this pattern and that this printer was printing fine before hand I do not think this is any kind of mechanical leveling or flatness issue. The print bed is 1/4" thick jig plate.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2703725/klippy.log)

[image]

ifUpEth0, 2018-12-21:
I started the same print job after doing a (BED_MESH_CLEAR) and a homing routine. 

[image]

Arksine, 2018-12-23:
This is possibly due to the new fade_target option which defaults to the average z position in the mesh.  Try setting fade_target: 0 in your printer.cfg to see if it behaves as you expect (this would mimic previous behavior).

KevinOConnor, 2019-01-10:
It looks like this was resolved, so I'm going to close this issue.

-Kevin

closed 2019-01-10 by KevinOConnor