Klipper3d/klipper #1027: Z axis step per mm out of range

Geekoid85, 2018-12-21:
Hello,
I have installed klipper on my raspberry pi 1 b and my home made reprap prusa i3. I sucess at configuring my pi and the xyz axis movments and end stop but I have a problem.

When I want to change the step distance of the z axis from 0.0025 to 0.00025 (witch is what my printer required, you know old hardware store lead screw :) klipper say : 
`Send: G28 XYZ
Recv: // Klipper state: ShutdownRecv: !! Internal error on command:"G28"
Recv: ok`
This happend even with .00025 instead of 0.00025
I think that I found software max step per millimiters limit !

Geekoid85, 2018-12-28:
[klippyLOG.zip](https://github.com/KevinOConnor/klipper/files/2715358/klippyLOG.zip)
Here they are !

jakep82, 2018-12-28:
You have a compounding problem of a very slow host computer, and a very small step distance.  The host needs to generate 800,000 steps to home Z, and a Pi 1b isn't fast enough to generate those steps in the time allotted.  Try the solution at the link.

https://github.com/KevinOConnor/klipper/issues/656#issuecomment-421860386

Geekoid85, 2018-12-28:
Okay thanks, this solution haven't fix my error so the next time I will setup klipper on a pi3. :+1:

closed 2018-12-28 by Geekoid85