Klipper3d/klipper #1031: BED_MESH_CALIBRATE doesn't do much

gcormier, 2018-12-22:
Issue a G28, homes. Issue BED_MESH_CALIBRATE, it homes to the first probe point then just sits there. Printer continues to respond to commands (ie. another G28).

`Stats 61328.4: gcodein=393 mcu: mcu_awake=0.019 mcu_task_avg=0.000140 mcu_task_stddev=0.000126 bytes_write=22874 bytes_read=25462 bytes_retransmit=9 bytes_invalid=0 send_seq=1358 receive_seq=1358 retransmit_seq=2 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993080 heater_bed: target=60 temp=59.6 pwm=0.309 print_time=61026.987 buffer_time=0.000 print_stall=0 extruder: target=190 temp=190.2 pwm=0.285
Stats 61329.4: gcodein=398 mcu: mcu_awake=0.019 mcu_task_avg=0.000140 mcu_task_stddev=0.000126 bytes_write=22973 bytes_read=25597 bytes_retransmit=9 bytes_invalid=0 send_seq=1366 receive_seq=1366 retransmit_seq=2 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993079 heater_bed: target=60 temp=59.7 pwm=0.256 print_time=61026.987 buffer_time=0.000 print_stall=0 extruder: target=190 temp=190.1 pwm=0.305
Dumping gcode input 50 blocks
Read 61226.354984: 'M105\n'
Read 61228.356895: 'M105\n'
Read 61230.358127: 'M105\n'
Read 61232.359742: 'M105\n'
Read 61234.361242: 'M105\n'
Read 61236.362600: 'M105\n'
Read 61238.364209: 'M105\n'
Read 61238.909023: 'bed_mesh_clear\n'
Read 61240.365052: 'M105\n'
Read 61240.555691: 'BED_MESH_CALIBRATE\n'
Read 61242.366790: 'M105\n'
Read 61244.368428: 'M105\n'
Read 61246.369635: 'M105\n'
Read 61248.370862: 'M105\n'
Read 61250.373171: 'M105\n'
Read 61252.374235: 'M105\n'
Read 61254.375484: 'M105\n'
Read 61256.376633: 'M105\n'
Read 61258.378332: 'M105\n'
Read 61260.379700: 'M105\n'
Read 61262.382422: 'M105\n'
Read 61264.386448: 'M105\n'
Read 61266.386925: 'M105\n'
Read 61268.388260: 'M105\n'
Read 61270.386824: 'M105\n'
Read 61272.388952: 'M105\n'
Read 61274.389995: 'M105\n'
Read 61276.391144: 'M105\n'
Read 61278.392252: 'M105\n'
Read 61280.394313: 'M105\n'
Read 61282.395430: 'M105\n'
Read 61284.396533: 'M105\n'
Read 61286.397621: 'M105\n'
Read 61288.399657: 'M105\n'
Read 61290.400747: 'M105\n'
Read 61292.401925: 'M105\n'
Read 61294.403101: 'M105\n'
Read 61296.404916: 'M105\n'
Read 61296.656692: 'BED_MESH_CALIBRATE\n'
Read 61298.405273: 'M105\n'
Read 61300.407252: 'M105\n'
Read 61301.894209: 'G28\n'
Read 61318.411123: 'M105\n'
Read 61320.412479: 'M105\n'
Read 61322.413451: 'M105\n'
Read 61323.298477: 'BED_MESH_CALIBRATE\n'
Read 61324.415505: 'M105\n'
Read 61326.416613: 'M105\n'
Read 61328.417468: 'M105\n'
Read 61330.360492: 'M112\n'
gcode state: absolutecoord=True absoluteextrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[25.0, 5.0, 4.0, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=8.33333333333
Shutdown due to M112 command
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Shutdown due to M112 command
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Stats 61330.4: gcodein=445 mcu: mcu_awake=0.019 mcu_task_avg=0.000140 mcu_task_stddev=0.000126 bytes_write=23058 bytes_read=25753 bytes_retransmit=9 bytes_invalid=0 send_seq=1374 receive_seq=1374 retransmit_seq=2 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993077 heater_bed: target=60 temp=59.9 pwm=0.000 print_time=61030.726 buffer_time=0.000 print_stall=0 extruder: target=190 temp=190.2 pwm=0.248
MCU 'mcu' shutdown: Command request
[... 184 lines cut ...]
Receive: 80 61327.944605 61327.942552 11: seq: 1c, clock clock=1495341395
Receive: 81 61328.005982 61327.942552 14: seq: 1c, analog_in_state oid=16 next_clock=1500983808 value=842
Receive: 82 61328.206749 61328.018698 14: seq: 1e, analog_in_state oid=6 next_clock=1504183808 value=6867
Receive: 83 61328.304493 61328.018698 14: seq: 1e, analog_in_state oid=16 next_clock=1505783808 value=841
Receive: 84 61328.506053 61328.018698 14: seq: 1e, analog_in_state oid=6 next_clock=1508983808 value=6856
Receive: 85 61328.607721 61328.556018 14: seq: 1f, analog_in_state oid=16 next_clock=1510583808 value=847
Receive: 86 61328.808464 61328.655625 14: seq: 10, analog_in_state oid=6 next_clock=1513783808 value=6863
Receive: 87 61328.906471 61328.856284 14: seq: 11, analog_in_state oid=16 next_clock=1515383808 value=845
Receive: 88 61328.932157 61328.926914 11: seq: 12, clock clock=1511084355
Receive: 89 61329.107740 61329.019279 14: seq: 14, analog_in_state oid=6 next_clock=1518583808 value=6865
Receive: 90 61329.205720 61329.156694 14: seq: 15, analog_in_state oid=16 next_clock=1520183808 value=843
Receive: 91 61329.406297 61329.256612 14: seq: 16, analog_in_state oid=6 next_clock=1523383808 value=6868
Receive: 92 61329.505472 61329.456285 14: seq: 17, analog_in_state oid=16 next_clock=1524983808 value=847
Receive: 93 61329.705976 61329.556372 14: seq: 18, analog_in_state oid=6 next_clock=1528183808 value=6860
Receive: 94 61329.807718 61329.756854 14: seq: 19, analog_in_state oid=16 next_clock=1529783808 value=841
Receive: 95 61329.914595 61329.910716 11: seq: 1b, clock clock=1526817786
Receive: 96 61330.008717 61329.910716 14: seq: 1b, analog_in_state oid=6 next_clock=1532983808 value=6853
Receive: 97 61330.106991 61330.019776 14: seq: 1d, analog_in_state oid=16 next_clock=1534583808 value=841
Receive: 98 61330.308498 61330.019776 14: seq: 1d, analog_in_state oid=6 next_clock=1537783808 value=6855
Receive: 99 61330.370179 61330.364914 12: seq: 1e, shutdown clock=1534082574 static_string_id=8
Stats 61331.4: gcodein=445 mcu: mcu_awake=0.013 mcu_task_avg=0.000114 mcu_task_stddev=0.000097 bytes_write=23143 bytes_read=25931 bytes_retransmit=9 bytes_invalid=0 send_seq=1381 receive_seq=1381 retransmit_seq=2 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993076 heater_bed: target=60 temp=59.7 pwm=0.166 print_time=61030.726 buffer_time=0.000 print_stall=0 extruder: target=190 temp=190.3 pwm=0.248
Stats 61332.4: gcodein=445 mcu: mcu_awake=0.013 mcu_task_avg=0.000114 mcu_task_stddev=0.000097 bytes_write=23222 bytes_read=26098 bytes_retransmit=9 bytes_invalid=0 send_seq=1387 receive_seq=1387 retransmit_seq=2 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993074 heater_bed: target=60 temp=59.8 pwm=0.058 print_time=61030.726 buffer_time=0.000 print_stall=0 extruder: target=190 temp=190.5 pwm=0.248
Stats 61333.4: gcodein=445 mcu: mcu_awake=0.013 mcu_task_avg=0.000114 mcu_task_stddev=0.000097 bytes_write=23308 bytes_read=26291 bytes_retransmit=9 bytes_invalid=0 send_seq=1394 receive_seq=1394 retransmit_seq=2 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993073 heater_bed: target=60 temp=59.8 pwm=0.042 print_time=61030.726 buffer_time=0.000 print_stall=0 extruder: target=190 temp=190.2 pwm=0.332
Stats 61334.4: gcodein=445 mcu: mcu_awake=0.013 mcu_task_avg=0.000114 mcu_task_stddev=0.000097 bytes_write=23406 bytes_read=26496 bytes_retransmit=9 bytes_invalid=0 send_seq=1402 receive_seq=1402 retransmit_seq=2 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993071 heater_bed: target=60 temp=59.7 pwm=0.202 print_time=61030.726 buffer_time=0.000 print_stall=0 extruder: target=190 temp=190.5 pwm=0.208
`

gcormier, 2018-12-22:
printer.cfg:
```
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar3
position_min: -13
position_endstop: -12
position_max: 210
homing_speed: 50

[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^ar14
position_min: -7
position_endstop: -6
position_max: 225
homing_speed: 50

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: ^ar18
position_endstop: -0.25
position_max: 220
position_min: -2
homing_speed: 3

[homing_override]
set_position_z: 5
gcode:
    G90
    G1 Z9 F5000
    G28 X0 Y0
    G0 X102 Y110 F5000
    G28 Z0
    G0 Z5 F500

[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
# Titan Extruder - 0.00249 measured with caliper
step_distance: 0.00249
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: ATC Semitec 104GT-2
sensor_pin: analog13
#control: pid
#pid_Kp: 22.2
#pid_Ki: 1.08
#pid_Kd: 106.889
min_temp: 15
max_temp: 250

# thermally controlled hotend fan
[heater_fan hotend_fan]
pin: ar5
max_power: 1.0
kick_start_time: 0.100
heater: extruder
heater_temp: 50.0
fan_speed: 1.0

[heater_bed]
heater_pin: ar11
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
# Limit power so the adhesive doesn't melt
max_power:0.8
#control: pid
#pid_Kp=63.832
#pid_Ki=3.404
#pid_Kd=299.213
min_temp: 15
max_temp: 110

# Print cooling fan
[fan]
pin: ar4

[display]
lcd_type: st7920
cs_pin: ar16
sclk_pin: ar23
sid_pin: ar17
encoder_pins: ^!ar31, ^!ar33
click_pin: ^!ar35

[bed_mesh]
speed: 50
samples: 2
horizontal_move_z: 4
sample_retract_dist: 2.0
min_point: 25,5
max_point: 203,153
fade_start: 1.0
fade_end: 2.0
probe_count: 8,8
algorithm: bicubic

[mcu]
serial: /dev/ttyACM0
pin_map: arduino

[printer]
kinematics: corexy
max_velocity: 300
max_accel: 3000
max_z_velocity: 25
max_z_accel: 30
square_corner_velocity: 5.0

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 23.080
#*# pid_ki = 0.850
#*# pid_kd = 156.656
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 44.530
#*# pid_ki = 0.486
#*# pid_kd = 1020.289
```

Arksine, 2018-12-23:
If you are manually probing,= you need to manually move the tool to the bed (using the paper test) and use the NEXT command to move between points.  You only get automatic leveling if you have a probe.

gcormier, 2018-12-23:
Ahh got it. Found the probe section in the examples extras file, configured, good to go :)

closed 2018-12-23 by gcormier