Klipper3d/klipper #1032: Best "board" for klipper?

swatti87, 2018-12-22:
Short story:

I need the best klipper-board there is. But what is it? 
2-extruders, 3-X fans... 

I need drivers that can move Nema23s and Nema17s
Currents set from config plz.

The long story:

My current setup is using Raspy, Arduino Due and custom board called Oulware. 

Problem is, i keep having "small problems" one after another and for some reason it allways ends up in something breaking down, usually the Raspy or the Due. 

Most recently one of my chinaasium fans started to have a wire loose. One of the hairs of that wire shorter the mosfet controlling the fans in the board. Things keep leading to more disasters so now i have about half-a-dozen Raspies and Dues burned up or broken, i dont have any means to tell if something is broken or not and TWO of the custom boards with the unknown status on em.

It is VERY unlikely that i can measure any readings from them without shorting something so "thats it for me then" - Well, atleast for now with this setup.

I need a new board. EVERYTHING!

I'l be building a new rig at some point, VERY big machine most likely using Nema23s so "lets make it a bit overkill" while we are at it. I'l need that later. 

I liked my TMC2130s because of their sensorless homing but im willing to let go of that if there is better drivers to be had.
I hate wiring, the Oulware is "2130 native" so it needs no extra wiring with the 2130s. The part that i liked about it.

What drivers can move big motors like N23s? Also, i'd prefer if i can set the currents from my config. I hate those tiny potentiometers. Just one more potential shortcircuit waiting to happen with me tuning currents. Lets eliminate potential hazards.

The board itself. What is the best "klipper board" nowdays?

swatti87, 2018-12-22:
I don't want to throw money at my problems but quite literally touching the board has caused something to short and there goes another raspy/due

I like these, hell, I love em when they work but I just can't trust them anymore. 
Anything, everything can cause an issue killing the whole setup, damaging 50e worth of parts, etc. 

The nastiest part is I don't know what happened most of the time. Something just broke and took another board with it...

I love the help I've been getting on these also but I feel bad about troubling people all the time. Also, they ain't online all the time.

I can't diagnose the problem on these, mostly because I don't know what caused it to begin with. 

All I know is the mosfets are dead. I know which ones too. 
One board has 3 working fets, one blown. Other board has 1 working and 3 blown. 

It connects to Octo and I can send commands but nothing moves or heats... 

Electronics is my arch-nemesis. I don't even know well how to measure stuff with a multimeter. Only thing I was able to was the PSU. 
I still think my end is the issue but my reading from the PSU was 3,7v and I tested 2 different PSUs... Or my meter is borked. I used a AAA battery to get a 1.4v reading so I'm even measuring it right. 

Could the PSU blow from this?

AlexSkarbo, 2018-12-22:
I think we need to make a list of supported microcomputers and microcontrollers, with some parameters, for example, that works only through a virtual card and with an expanded transfer buffer, there are problems of compatibility with Uart speed, so that people would know what to buy and what to pay for. How do you think?

jakep82, 2018-12-22:
If you aren't comfortable with a multimeter, then I don't think a hand made, custom designed PCB is the right choice for you.  Like I said, I had to replace both gate drives on mine and add a couple resistors and capacitors to prevent them from blowing again.  That might be what's wrong with yours since the gate drivers are what control the fets.  I actually have an extra gate driver looking for a home, so like I said, I would be happy to buy your boards and you can move on to a commercial board with better support.  The Duet Wifi is your best bet if you want excellent hardware, and proper support.  Of course it's meant to run RRF, so you might find limited support for running Klipper on it.

Regarding voltage, I'm very confused by what you're saying.  I use a 24V PSU for the motors and heaters.  I then use buck converters to provide 5V and 12V.  Where are you seeing 3.7V?

swatti87, 2018-12-23:
I can do some soldering, i swapped the fets, proud moment. Its just very far from my comfort-zone. 

I'd need a board with proper support. 

As for the PSU. I just poked the black to the black wire and red to red on the multimeter. Reading for both is ~3.7v
Doing the same to the AAA-battery gives 1.4v
I was under the impression it should be at the 24v...?

jakep82, 2018-12-23:
This isn't really the right place to have a conversation about hardware problems.  I'm on the Klipper IRC, mailing list, and unofficial Discord if you want to continue discussing hardware.

swatti87, 2018-12-23:
I dont really know how all of these work, or should work so lets eliminate a problem if we can, without me having to use dozen different platforms to do that.

I have a fairly cheapo multimeter, could that be the cause? I need to find something that runs on 24v and measure that, see if it gives my meter the same reading.
Should the PSU give the multimeter a 24v reading?

I was under the impression MeanWell PSUs are quite tough. Surely they have a fuse or something to protect them? I might have shorted the wires more than once before and it was still kicking. I bought a second, "used" bigger PSU that is now connected and as mentioned, it gives the same reading. I havent used it or tested it before but i doubt the PSU is to blame here unless the other PSU was a lemon to begin with... 

And then software: 
Is there a code to test out if the Due is dead?

Several logs point to "Serial-port error" and the octo terminal reads this: "Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data"

https://faq.octoprint.org/serialerror - Went there, double checked everything.
All wiring is fine, none have been touched/changed. Raspy PSU is the original meant-for-raspy psu. 
There is nothing else connected to the raspy USB ports. Just the cable going to the Due prog-port.

jakep82, 2018-12-26:
If you're constantly blowing up Pis, Arduinos, and components on control boards, that would lead me to suspect something else is at fault.

Regarding speed, 200mm/s in a typical printer configuration is only 16K steps per second for a single axis.  An AVR can easily handle that, even if you factor in all the other steppers.

jdew, 2019-01-04:
I'm half tempted to order an omega2+ onion with their R2 arduino dock with some form of ramps board and give it a go.

KevinOConnor, 2019-01-16:
I'm going to close this issue as I don't think we'll ever have an exact answer to the original question.

-Kevin

closed 2019-01-16 by KevinOConnor