Klipper3d/klipper #1033: Klipper to uStepper using BUS I2C/SPI (no central controlboard)

FixFuture, 2018-12-22:
Hi!
I have been researching posiibility of building 3d-printer without centralized control board (like duet wifi) so that
1) Rpi runs octoprint and klipper
2) each device would connected via bus (i2c/spi/...)
3) steppers would be smart st

KevinOConnor, 2018-12-24:
FYI, I've been working on adding support for the Mechaduino (and the very similar "Smart Stepper" and their clones) to Klipper.  More details at PR #1038.

The support I'm working on uses the USB bus for all communication.  If someone wishes to add support for SPI/I2C then that's fine, but that isn't something that interests me.

-Kevin

FixFuture, 2018-12-28:
Great! 

I'll take look at this. I have notified uStepper development team about klipper. 

The problem that I see with USB implementation is that number od steppers is limited to number of usb ports available and require separate power source.

I'll see if my skills are enough to contribute.

KevinOConnor, 2019-01-10:
Sounds good.  I'm going to close this issue.  Further updates on mechaduino support can be tracked on PR #1038.

-Kevin

closed 2019-01-10 by KevinOConnor