Klipper3d/klipper #1037: prioritize SET_GCODE_OFFSET Z_ADJUST commands during a print

pecirep, 2018-12-24:
Sending a gcode command via octoprint puts it in the gcode queue on the mcu, everything that was sent before will be executed first. With things like SET_GCODE_OFFSET Z_ADJUST=-0.2 for example (if your printhead is too high and filament isn't sticking) this can be a problem, as a square skirt at the start of the print will finish the current edge, sometimes even the next edge, before the z offset is adjusted. Is there any way to prioritize a gcode command so it is executed immediately or at least before the next gcode command in the queue? In my case the Z motors are on a separate arduino, which isn't doing anything during XYE moves, maybe there's a way to directly send a klipper-format command to it?

rmcbc, 2018-12-25:
I made same changes in klipper and a pull request to try to solve your problem https://github.com/KevinOConnor/klipper/pull/1039

KevinOConnor, 2019-01-10:
I'm inclined to close this issue as I don't see a short term resolution for it.  I don't know of any firmware that applies a gcode offset out of order.  I can understand the concern that Klipper has a longer look-ahead queue (typically 2 seconds) than most other firmware, but I don't see a way to reasonably address that (short of reducing the lookahead queue time, but that has other issues).

-Kevin

pecirep, 2019-01-11:
My suggestion would have been to make it an option. If the offset command has a certain parameter (for example FAST, NOW, INSTANT) it gets executed not just out of order, but even during the current command if need be. (I've seen the command take up to 15 seconds to be executed because long straight lines were being printed) Something like this would also be useful for M109 and M190, as there is currently no way to cancel heatup other than triggering E-Stop (which shouldn't be the goto in my opinion).

But I do understand if you consider these unfeasable, unsafe or simply not a priority. In that case the issue can be closed, this was merely a suggestion.

talldonkey, 2019-01-14:
This exact delayed babystepping issue is what frustrated me enough to go back to Marlin.  At first I didn't understand it would be a queue, so kept hitting the babystep and wondering why it wasnt working.... commands eventually came up in queue, boom, nozzle buried in bed :)

Would temporarily lowering the lookahead queue enough to give the appearance of instantaneous babystep response just for the first layer be feasible?  Then for the rest of the print, it goes back to normal lookahead.  The thought is that its just the first layer that we use babystepping and want immediate response to babystep adjustments.

KevinOConnor, 2019-01-14:
It's possible to lower the lookahead time.  Lower it too much, though, and it will lead to "timer too close" errors.  I don't know what the minimum safe lookahead time is.  Last time I looked at this (a couple of years ago) I felt 2 seconds was a safe trade-off between getting good queuing and having responsiveness to new commands.

-Kevin

drphil3d, 2019-01-18:
even in marlin it can sometimes take a second or two for the offset adjustment to change while printing. It's usually not something that has to be adjusted a lot.

drphil3d, 2019-01-19:
Using Set gcode offset adjust during printing doesn’t seem to change anything. The z offset in my config file doesn’t change either. Am I doing something wrong ?

pecirep, 2019-01-26:
Kevin, your reasonning for not wanting to change this functionality make sense to me now, so I'll close this issue

closed 2019-01-26 by pecirep