Klipper3d/klipper #1040: Internal error on command:"G28" MCU 'mcu' shutdown: Timer too close

gti96, 2018-12-25:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2708895/klippy.zip)

Using CTC prusa i3 pro b with octopi.

I am having problem calibrating z-stepper step distance.  I believe I need to set it to 0.004 from default 0.05. This happens when homing z-axis using G28. With step distance 0.05 and homing speed of 100 it works. Tried step distance of 0.025 and 5 homing speed and it works. Lowest I could get working is 0.009 with 5 homing speed. Anything lower than 0.009 would not work. Max z-velocity and acceleration has been reduced to 5 and 30. 

Please help.

gti96, 2018-12-25:
Just tested again and 0.009 with 5 doesn't work. it seems intermittent. Before I could get to 0.009, it seems it was stuck at 0.016 at lowest for quite some time.

gti96, 2018-12-25:
also changed controller board to MKS Gen-L v1.0 with A4988 drivers

gti96, 2018-12-25:
with RB Step Motor 17HDC4023-24B

ndupont-net, 2018-12-26:
You are trying to generate more steps than what your MCU is able to produce.

We are talking 100k steps/sec on the Arduino 2560.  You will need to lower the Z speed in order to add resolution.

jakep82, 2018-12-26:
The steps look fine to me, even on a 2560.  In the log I see a step distance of .0004, and at a homing speed of 5mm/s we're only talking about 12,500 steps per second.  What type of Pi are you using to run Klipper?  If it's a Pi 0,1, or 2, you'll need to change the step delay in homing.py to get around this issue.  See the link for more info.

https://github.com/KevinOConnor/klipper/issues/773#issuecomment-431595545

ndupont-net, 2018-12-26:
Oops my bad... I actually run my Tarantula with step_distance: .000627408 and max_z_velocity: 25 with no issue (MKS Base 1.4). It was fine even on the Pi Zero W.

jakep82, 2018-12-26:
> Oops my bad... I actually run my Tarantula with step_distance: .000627408 and max_z_velocity: 25 with no issue (MKS Base 1.4). It was fine even on the Pi Zero W.

The difference is likely related to the overall Z height.  This printer has a 400mm Z, so Klippy needs to generate 1,000,000 steps to home Z.  A slower host can't generate that many steps in the time allotted, so it shuts down with an error.

gti96, 2018-12-27:
I am sorry. I am relatively new to 3d printing. My printer is actually 200mm Z. I changed it to 400mm Z so that I could move the z further up physically before my calibration is right. 

Currently a Pi 1B+ is being used I believe so I have tried adjusting step delay to 0.000001 and it's still the same. then I tried 0.00005 and it is working with step distance of 0.0004 and 5mm/s. Thank you. 

What does this step delay do? and is this in seconds? what would be the optimal step delay in my case?

gti96, 2018-12-27:
Sorry, correction. I tried 0.000001 again and it works now. weird...