Klipper3d/klipper #1042: BL Touch Clone not working [log, config and wiring included]

welo78, 2018-12-25:
Hello, I have been having problems with succesfully homing/implementing a working BL Touch clone. The problem is that the probe itself is working, the self test was correct. But when trying to test it via DEBUGGING command it does not seem to be switching between opened/triggered I tried with pullups(^) and with negatives (!) and the combination. Nothing worked.

I am using a MKS Gen 1.4 board with BL Touch setup as this:
[Wiring BL touch](http://3dprintfaq.com/images/MKS-BASE_GenV1.4_BLTouch.jpg)

Expected behaviour: Succesfull BL Tuch test and homing
Experienced behaviour: No change in open/triggered

Klipper log from testing the sensor:
[bltouch test log](https://github.com/welo78/Klipper/blob/master/klipper_fail_touch_log)

Commands used:
Send: BLTOUCH_DEBUG COMMAND=pin_down
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_down
Send: QUERY_PROBE
Recv: // probe: TRIGGERED

Send: BLTOUCH_DEBUG COMMAND=pin_up
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_up
Send: QUERY_PROBE
Recv: // probe: TRIGGERED

big-g, 2018-12-27:
@fredizzimo ,
Thanks for the hint on the undocumented `test_sensor_pin`. Works perfectly for my TLTouch on my Tevo Tornado.

cake808, 2018-12-31:
@fredizzimo ,
Could you elaborate what exactly you need to do to disable the `test_sensor_pin` option?
Thank you

fredizzimo, 2019-01-01:
@cake808,

Just add `test_sensor_pin: False` to the BLTouch section in your configuration. There should be no harm in disabling that, since the test does almost nothing in practice, it basically does this:

"If the probe behaves normally or there is an error (not a typo), then continue with the probing..." Note that @KevinOConnor and I have very differing views on that and my analysis is described in much more detail here https://github.com/KevinOConnor/klipper/issues/955#issuecomment-448260820

My pull request on the other hand, checks for the error state reported by the probe itself, so it should cover a lot of more situations. But note that it do have some problems, as can be seen in the last few comments of #955. At this time I don't know if @KevinOConnor's version has the same issue or not.

xsasx, 2019-01-07:
How can i update to your pull request ?

fredizzimo, 2019-01-08:
@xsasx 
https://help.github.com/articles/checking-out-pull-requests-locally/

One thing that I forgot to mention though, is that my pull request is not rebased on the latest version anymore. It's based on a very slightly older version of Klipper, so it might miss some of the very latest changes. You could rebase it yourself though.

However now that @KevinOConnor has opened https://github.com/KevinOConnor/klipper/pull/1091, I will probably close mine. But not yet, since I just reviewed his pull request and IMO there are a few critical things missing still.

KevinOConnor, 2019-01-21:
I'm going to close this issue as a number of changes have occurred to the bltouch code since the original report.

-Kevin

closed 2019-01-21 by KevinOConnor