Klipper3d/klipper #1049: Internal error on "PROBE" command.

mitry1974, 2018-12-28:
A try to setup new delta printer with klipper software.
When I try to probe I get error "Recv: // Klipper state: Shutdown
Recv: !! Internal error on command:"PROBE""
[klippy.log](https://github.com/KevinOConnor/klipper/files/2713917/klippy.log)

KevinOConnor, 2018-12-30:
Set a realistic minimum_z_position, and move the head to within about 20mm of the bed before issuing a PROBE command.  (The current Klipper code needs to generate and store the entire move in the micro-controller memory prior to starting a homing or probing move - using unrealistic values stresses this system.)

-Kevin

mitry1974, 2018-12-31:
Hello, thank you for the answer. 
I try to switch drivers to 64 microsteps and after that everything is Ok - I can do probe and delta_calibrate commands. When I try to set 128 microsteps I have the same error. I use lv8729 drivers....

KevinOConnor, 2019-01-10:
I'm going to close this issue as I think the original question was answered.

-Kevin

closed 2019-01-10 by KevinOConnor