Klipper3d/klipper #1050: Really hate to ask - update using specific code (Manual Mesh, Menu, etc)

Spartacus1066, 2018-12-28:
It looks like mcmatrix has some work in this tree 1fb8fe48c92c7a1972f1da507fef20a410e921b6 that I would like to try. It's not in master yet, right?
Sorry, just feeling my way around - I promise I searched and read for 2 hours before asking!

I am just trying to get manual mesh going.
Thanks
https://github.com/KevinOConnor/klipper/tree/1fb8fe48c92c7a1972f1da507fef20a410e921b6/

KevinOConnor, 2019-01-06:
I'm not sure what's being asked here.  We are working to get the probe menu code into the main klipper repo.

-Kevin

Spartacus1066, 2019-01-06:
I was asking if I was able to merge his in-progress code (other than copy/paste). I did not know if there was a "clean" way to test that code and revert or whatever. Thank you, sir!

KevinOConnor, 2019-01-16:
You should be able to navigate to the repository owner and then checkout the specific branch from that repository.  There should also be some guides online for doing this.  I'm not familiar with the exact details.

-Kevin

closed 2019-01-25 by KevinOConnor