Klipper3d/klipper #1052: filament selection: Multi toolhead machine? or extending existing beyond XYZE?

trevjonez, 2018-12-28:
So now that I can at least get force move to make the mmu2 perform basic moves I want to start laying out more robust ways of control than depending on diagnostic options such as force_move.py.

So far all current approaches have failed at even getting basic homing behaviors to work due to the hard constraint of homing.

any thoughts on how to best keep code reuse high and add the ability to extend the axis set beyond the typical XYZE?

current attempt has a config that looks like this:
```
[...]
[printer]
kinematics: selector
max_velocity: 200
max_accel: 1250
max_z_velocity: 12
max_z_accel: 200

[kinematics]
toolhead: cartesian
filament: ips
```

selector kinematics being a composite root for the kinematics that then loads up and delegates to the toolhead and filament objects.

I am sure I could keep blindly hacking but I would rather see if anyone else has gone into this sort of thought flow first?

as for the actual components and naming scheme. you can pretty easily see it by skimming through the [assembly guide here. ](https://manual.prusa3d.com/Guide/2.+Pulley+body+assembly/755?lang=en)

basically we have the idler drum/tumbler that would home and have a preset list of positions at specific angles from home. then similarly the selector axis would have a preset list of positions at specific distances from home. and finally a pulley axis would behave as an extension of an extruder axi. if the user were to take time to calibrate the pulley step distances to match that of a correctly calibrated extruder it could in effect be used in tandem to drive and retract filament during filament changes.

then to make things slightly more interesting the selector axis might be constrained by filament making homing impossible at which point a user should be able to inform the host which slot is currently selected thus making the act of finding home unnecessary as the location is known.

KevinOConnor, 2019-01-02:
Interesting.  Out of curiosity, could this code be implemented mostly in a new klippy/extras/filament_selector.py module?  That is, could the configuration be done in a [filament_selector] config section, have it add new commands (eg, FILAMENT_SELECTOR_HOME) and have it register new virtual extruders (so that T1/T2/etc automatically performed the necessary filament change sequences)?

-Kevin

trevjonez, 2019-01-02:
yea I think that would be the ideal solution. my current deal was just me trying to short circuit getting into the existing homing flow. I think the [filament_selector] config is probably the best route to take. I just wonder what the best way to attach it to the toolhead flow is.

I think my next best bet will be to literally draw out the bigger code flow on paper so I can better understand the relationships.

Then as always I want to make sure to keep the filament_selector extensible so that it can be extended for things like the mmu1 (many extruders feeding a many to one ptfe reducer.) or the s3l3ctor which is sort of a mix of mmu1 and mmu2 where you have two steppers for idler and pulley feeding a similar many to one reducer.

KevinOConnor, 2019-01-02:
> I just wonder what the best way to attach it to the toolhead flow is.

Does it need to be attached to the toolhead?  I thought the mmu2 was only active on a filament change (eg, on a T0/T1 command).  If it's only active on a filament change, then a filament_selector.py module should be able to register an extruder class, and then perform the necessary filament switching work when that class' set_activate() method is invoked.

> Then as always I want to make sure to keep the filament_selector extensible so that it can be extended 

It might be simpler to introduce independent mmu2_filament_selector.py and mmu1_filament_selector.py modules.  The python code can still use "import" to pull in code and reuse common classes - for an example of this see how bltouch.py imports probe.py.  That said, I don't know enough about the mmu1, mmu2, or s3l3ctor to know if that makes sense or not.

-Kevin

trevjonez, 2019-01-02:
You are on the right thought path. Only reason I would want the toolhead hooked in is because there needs to be some overlap with the filament selector and extruder moves as well as filament sensors to help avoid the current headache people are seeing with the prusa firmware impl. certainly though that can be added after the fact. I'll do some more studying on the way homing works to figure out what the right solution is. I'll reach back out once I have a better understanding of the existing code bits.

KevinOConnor, 2019-01-21:
I'm inclined to close this issue unless there are further questions.

-Kevin

trevjonez, 2019-01-21:
I do have a question in regards to the axi tracking. Currently the toolhead kinematics etc is coupled to XYZE via the size four float array. I have a local branch where I started modifying that to be a variable length array of objects so that additional axi's could be included with the same requested print time for move calculation/queuing. Looking at cmd_GET_POSITION in gcode it seems we have `mcu_pos`, `stepper_pos`,  `kinematic_pos`, `toolhead_pos`, `gcode_pos`, 'base_pos` and `homing_pos`. 

It seems as through kinematic pos and toolhead_pos should be merged into just `kinematic_pos`?

After that the distinction of gcode_pos vs base_pos vs homing_pos is a bit fuzzy to me. 

gcode pos seems to be a copy of toolhead_pos in order to help with relative command math? 

base_pos would be a modifier(gcode offset) on top of homing_pos? which I assume should be set by the kinematics during homing(or force move set position) meaning that a home pos is more a property of an axis which is state that would be owned by kinematics? 

Finally in what cases would you expect mcu_pos and stepper_pos to be different? or are they one in the same?

KevinOConnor, 2019-01-21:
> Currently the toolhead kinematics etc is coupled to XYZE via the size four float array.

The high-level toolhead uses an XYZE 4-tuple and the low-level kinematics code (as found in klippy/chelper/) uses an XYZ "struct coord".  I think it would be interesting to extend this to six degrees of freedom (XYZUVW), but I suspect that isn't what you are looking at.

I wonder if modifying the toolhead code is the right approach for the MMU2.  The goal of the toolhead.py code is to track the location of the main printing "tool".  The MMU2 does not physically move with the main "tool" effector, and so it seems odd to track it's location there.  I would have thought that tracking the MMU2 state would be better in a separate class dedicated to it's "tool" locations.

> It seems as through kinematic pos and toolhead_pos should be merged into just kinematic_pos?

The toolhead_pos is the last location that the tool effector was commanded to move to.  It is different from kinematic_pos in a couple of key ways:
1. The toolhead position is the last commanded position of moves entering the lookahead queue, while the kinematic position is the position last sent to the micro-controller (ie, of the last move to exit the lookahead queue).
2. The commanded position may different from the stepper positions due to rounding.  For example, it's possible to command the effector location to an X position of 1.999999232341 and toolhead_pos will track that.  In contrast, the kinematic position is a calculation of the cartesian coordinate of the effector from the current stepper locations (which have taken an discrete number of finite steps).  Thus on a cartesian printer with step_distance of .010 then one would expect a kinematic position of X=1.990 or X=2.000.

> After that the distinction of gcode_pos vs base_pos vs homing_pos is a bit fuzzy to me.

This is the insanity that is g-code - nothing is ever expressed in a sane coordinate system.

> gcode pos seems to be a copy of toolhead_pos in order to help with relative command math?

The gcode pos can be different from toolhead pos if bed_mesh and/or bed_tilt are in effect.  The toolhead always works on a static coordinate system - the bed translations take the requested g-code moves and alter them to the desired toolhead coordinate system.

> base_pos would be a modifier(gcode offset) on top of homing_pos?

base_pos is not a "position" but an axis location.  It differs from 0,0,0,0 if a G92 command is used to alter the meaning of g-code commands.

homing_pos is similar - it is also an axis location.  It differs from 0,0,0,0 if SET_GCODE_OFFSET of M206 is used to alter the meaning of g-code commands.  It primarily differs from base_pos in that it is not reset after a G28 homing command.

> which I assume should be set by the kinematics during homing(or force move set position) meaning that a home pos is more a property of an axis which is state that would be owned by kinematics?

I'm not sure what you are asking here.  FORCE_MOVE doesn't maintain positional state - it blindly moves the stepper without informing the other parts of the system.  It is intended as a debug command.

For other types of movements not directly initiated from g-code, the preferred approach is to inform the toolhead class what the position of the tool effector is actually at, and then to run gcode.reset_last_position() so that it can reset its state accordingly.  Again, though, I wouldn't think this would be needed for the MMU2 as it should be possible to track its state separately from the location of the main tool effector.

> Finally in what cases would you expect mcu_pos and stepper_pos to be different? or are they one in the same?

The stepper_pos is a floating point number containing the commanded position (at the end of the lookahead queue - similar to kinematic position) of the stepper.  The mcu_pos is an integer number containing the total number of steps taken on the stepper since the micro-controller was last reset.  It should be possible to calculate mcu_pos from stepper_pos (using the stepper step_distance and an offset).

-Kevin

trevjonez, 2019-01-21:
excellent thank you sir. Main reason I want to extend toolhead to be aware of the MMU2 is so that I can synchronize move commands between E and possible other extruder motors. I think we can indeed close this for now.

closed 2019-01-21 by trevjonez

KevinOConnor, 2019-01-21:
> Main reason I want to extend toolhead to be aware of the MMU2 is so that I can synchronize move commands between E and possible other extruder motors.

This would indeed be useful, and I do feel it would be useful to extend the code to improve this type of synchronization.

FYI, I have been considering changing the toolhead code so that the kinematic classes register all interested steppers with the toolhead class.  Then, on each move, every stepper can itself determine what steps need to be issued to accomplish the given movement.

This could be useful for improvements to pressure advance, possible improvements to reduce ringing, support for "mixing" extruders, etc.

In the above organization, an mmu2 class could potentially register itself with the toolhead class and then arrange for extruder steps to be generated with each "e" movement of the main toolhead.

However, I don't have any timeframe for this type of change.  It's also unclear to me what the best way to accomplish this would be.

-Kevin