Klipper3d/klipper #1054: Config error for TMC2208 on Fysetc F6

SimoneBnc, 2018-12-30:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2717349/klippy.log)

I never used tmc2208 before so maybe it is my fault but I can't figure out the error....

KevinOConnor, 2018-12-30:
You'll need to define both a [stepper_x] section and a [tmc2208 stepper_x] section.  (Yeah, it's a little confusing - something for us to look at changing in the future.)  Take a look at config/example.cfg for the fields that go in stepper_x and look at config/example-extras.cfg for the fields that go in "tmc2208 stepper_x".

-Kevin

SimoneBnc, 2018-12-31:
I figured that out. Now I am getting this 
Starting serial connect
got {'#receive_time': 2051.751356144, u'next_clock': 2164480000L, u'oid': 8, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2051.723432029}
got {'#receive_time': 2051.861732238, u'next_clock': 2166240000L, u'oid': 19, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2051.824606196}
got {'#receive_time': 2052.0511130190002, u'next_clock': 2169280000L, u'oid': 8, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2052.0268650499997}
Loaded MCU 'mcu' 77 commands (v0.7.0-13-ge4e0fec-20181230_164416-OctoPi / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'mcu' config: RECEIVE_WINDOW=192 SOFT_PWM_MAX=256 SERIAL_BAUD=250000 ADC_MAX=1023 PWM_MAX=255 CLOCK_FREQ=16000000 MCU=atmega2560 STATS_SUMSQ_BASE=256
MCU error during connect
Traceback (most recent call last):
  File "/home/simone/klipper/klippy/klippy.py", line 140, in _connect
    cb('connect')
  File "/home/simone/klipper/klippy/mcu.py", line 769, in printer_state
    self._connect()
  File "/home/simone/klipper/klippy/mcu.py", line 624, in _connect
    self._check_config()
  File "/home/simone/klipper/klippy/mcu.py", line 568, in _check_config
    config_params = self._send_get_config()
  File "/home/simone/klipper/klippy/mcu.py", line 565, in _send_get_config
    self._name,))
error: Can not update MCU 'mcu' config as it is shutdown
got {'#receive_time': 2052.161507081, u'next_clock': 2171040000L, u'oid': 19, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2052.078646092}
got {'#receive_time': 2052.35164229, u'next_clock': 2174080000L, u'oid': 8, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2052.078646092}
got {'#receive_time': 2052.461752706, u'next_clock': 2175840000L, u'oid': 19, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2052.078646092}
got {'#receive_time': 2052.650704477, u'next_clock': 2178880000L, u'oid': 8, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2052.078646092}
got {'#receive_time': 2052.760823539, u'next_clock': 2180640000L, u'oid': 19, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2052.078646092}
got {'#receive_time': 2052.950939737, u'next_clock': 2183680000L, u'oid': 8, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2052.078646092}
got {'#receive_time': 2053.061063435, u'next_clock': 2185440000L, u'oid': 19, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2052.078646092}
got {'#receive_time': 2053.250048643, u'next_clock': 2188480000L, u'oid': 8, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2053.063674789}
got {'#receive_time': 2053.36018755, u'next_clock': 2190240000L, u'oid': 19, u'value': 8184, '#name': u'analog_in_state', '#sent_time': 2053.063674789}

SimoneBnc, 2019-01-02:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2720511/klippy.log)
Here is the latest log. I have tried with and without tx_pin but I still get the same. If a swap any of the drivers with one that hasn't the pads soldered together I get the message that it can't read register for that driver, so I think I am on the right path.

KevinOConnor, 2019-01-02:
The root cause of the error in that log file is an "adc out of range" error.  That's usually the result of an incorrect thermistor pin configuration (or a damaged / not wired thermistor).

-Kevin

SimoneBnc, 2019-01-02:
Ok, that worked thanks. I had thermistor unplugged as I was doing tests to have less cables in the way, I would have never thought it could prevent klipper from starting. Maybe I more user friendly error message would be helpfull since I was using a config file from another user with the same board.

JamieAKing, 2019-01-05:
@SimoneBnc I know this probably isn't the place to ask this but ....
How are you finding the Fysetc F6 board. I am currently building a new machine and am thinking of using this board coupled to a pi running clipper. 
my only concern is that I can find very few reviews of the board. for the money it seems to be to good to be true

SimoneBnc, 2019-01-06:
Besides the fact that I still haven't figured out how to get the ender 3 screen to work (it was in marlin), the board is really nice: uart for tmc is already working just move a jumper and close 2 solder pads on the drivers, all fans are pwm controlled and mosfets don't even get warm. If you don't really need 6 drivers I think you can get away with an mks that's a little cheaper. The other downside is the cabling as I had to redo all the connectors on the control board.

JamieAKing, 2019-01-06:
Thank you for the quick reply
I do need all the motors as I’m going to use a 3 motor self levelling z assembly. Does the pin out in the wiki page match up with the board when used in klipper config? When you say redo all the connectors is that just because it is clip in connectors as opposed to screw terminals?

SimoneBnc, 2019-01-06:
So far yes, I haven't used the last 2 drivers so I don't have mapped the pins yet as you need to use the hardware pin numbers as klipper supports only Arduino pin numbers of the standard mega2560 board while the F6 has an extended number. The F6 uses clip in terminals while the original board uses jst connectors, so they are a bit different; screw terminals are still there for all the heaters output and the 2 power inputs.

KevinOConnor, 2019-01-07:
> Besides the fact that I still haven't figured out how to get the ender 3 screen to work

FYI, an example was just added to config/generic-ramps.cfg for that (commit bd410c22).

> The other downside is the cabling as I had to redo all the connectors on the control board.

Out of curiosity, when you say you had to "redo all the connectors", do you mean you had to resolder something or do you mean you had to play "tic-tac-toe" with the jumpers to get the tmc uart pins routed?

-Kevin

SimoneBnc, 2019-01-07:
I'll try those pins.
I had to redo all the connectors for stepper motors, thermistors and fans. Connecting uart of the drivers it was just a matter of switching a jumper and soldering the 2 pads next to the driver's chip, no flying wires needed.

SimoneBnc, 2019-01-07:
Thanks a lot. Now it is working, 
For reference this is the display section

# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
[display]
lcd_type: st7920
cs_pin: PA5
sclk_pin: PA3
sid_pin: PA7
encoder_pins: ^PA1, ^PH0
click_pin: ^!PC2

If you want I can share the cfg file working on the F6.

JamieAKing, 2019-01-08:
That would be amazingly helpful

JamieAKing, 2019-01-08:
That’s sounding good on about the print quality. I’m sure it will get even better with a little more tweaking.
Please excuse my ignorance but this is not at all what I was imagining the config would look like! 
Where did you get the PIN numbers from? 
Once again I am completely in the dark (thought I understood how this all worked, that bubble has burst!) what are the pin abbreviations short for and how do the correspond to the pin map on the f6 wiki page?
I ask as I would like to remap two of the extruder drivers to drive z to give me 3 point levelling on my print bed   
Many thanks in advance

JamieAKing, 2019-01-09:
My machine is a full custom build, it is a Core XY with a lifting build plate form. When I probe the bed I will do so at the 3 points closest to the lead screw (each lead screw is paired with a linear rail for stability) then software can calculate the differences and move each motor independently so level the bed. The configuration of the lead screws allows for a flat level plane with no twist.

JamieAKing, 2019-01-09:
Got it, I see how it works now thank you 
What a ball ache converting it all. Still you have done all the hard work. Thank you again for all your help. Only ordered the board yesterday from China so going to have a little wait before it arrives....

SimoneBnc, 2019-01-09:
Let us know how it works

KevinOConnor, 2019-01-16:
I'm going to close this as I think the original question has been answered.

-Kevin

closed 2019-01-16 by KevinOConnor