Klipper3d/klipper #1056: Multiple none pins

ramezquitao, 2018-12-30:
I'm using klipper with replicape, and in my setup I have two spi temperature sensors. Right now it is possible to connect to each sensor individually by declaring the sensor pin as `sensor_pin: host:None`, but when I try to connect to both sensors at the same time it is not possible, because, two pins can not be named both `None`. In my setuo I changed bus.py line 64 from:

`if pin == 'None':`

to 

`if pin[:4] == 'None':`

So I can name one pin None, and the next None1. What would be the correct klipper-way
 to solve this issue?

By the way Happy New Year!!!

KevinOConnor, 2018-12-30:
That looks like a bug.  The shared pin error detection should have been disabled for "None".  I'd do something like the below (totally untested).

Happy New Year!!
-Kevin

```diff
diff --git a/klippy/extras/bus.py b/klippy/extras/bus.py
index c7e270ca..9721a591 100644
--- a/klippy/extras/bus.py
+++ b/klippy/extras/bus.py
@@ -62,6 +62,7 @@ def MCU_SPI_from_config(config, mode, pin_option="cs_pin",
     cs_pin_params = ppins.lookup_pin(cs_pin)
     pin = cs_pin_params['pin']
     if pin == 'None':
+        ppins.reset_pin_sharing(cs_pin_params)
         pin = None
     # Load bus parameters
     speed = config.getint('spi_speed', default_speed, minval=100000)
diff --git a/klippy/pins.py b/klippy/pins.py
index b424c4b9..0d168c55 100644
--- a/klippy/pins.py
+++ b/klippy/pins.py
@@ -252,6 +252,9 @@ class PrinterPins:
         can_pullup = pin_type in ['endstop']
         pin_params = self.lookup_pin(pin_desc, can_invert, can_pullup)
         return pin_params['chip'].setup_pin(pin_type, pin_params)
+    def reset_pin_sharing(self, pin_params):
+        share_name = "%s:%s" % (pin_params['chip'], pin_params['pin'])
+        del self.active_pins[share_name]
     def register_chip(self, chip_name, chip):
         chip_name = chip_name.strip()
         if chip_name in self.chips:
```

KevinOConnor, 2019-01-02:
I committed this change (commit 1be24b8f).

-Kevin

closed 2019-01-16 by KevinOConnor