Klipper3d/klipper #1057: Noob here, trying to use the LCD

ltmksh, 2018-12-31:
Hello!
I'm very new to Klipper world (2 days so far) and I must tell, it's great!
I could set it up to my needs (printer, drivers, custom G code), the only thing I miss is a more sofisticated menu on the LCD. 
While trying the menu points I found that "load filament" does not works as it should. 
By default it's:
G1 E200 F1000
G1 E100 F300
It feeds 200mm at higher speed then retracts 100mm at slower speed. 

I modified it to:
G1 E200 F1000
G92 E0 <--- reset extruder
G1 E100 F300

Now it works as it should.

Master92, 2019-01-05:
Just to make sure, I get the point: When you boot your printer from power off and start klipper right after that - does it behave this way or is it only after you did some printing before? I don't have my printer atm, because otherwise I would check it myself. What release are you using? v0.7.0?

KevinOConnor, 2019-01-06:
It does look like the default "load filament" menu should be changed.

-Kevin