Klipper3d/klipper #1059: I have to run my Ender 3 at 240C for PLA

peterlpop, 2018-12-31:
It used to be 210F by subjective quality. How do I calibrate the settings it to show about correct? (and then I will use an external probe to make it accurate_).

jakep82, 2018-12-31:
It sounds like you don't have the correct thermistor configured in your printer.cfg file.

peterlpop, 2018-12-31:
I have NTC Thermistor 100K 3950 (been looking for answers ... found the way to define a new one in config file). 

What do I put in my config file based on this model?

jakep82, 2018-12-31:
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L150

peterlpop, 2019-01-01:
Thanks. I guess using an already made ender3.cfg made me skip reading the example.cfg file.

KevinOConnor, 2019-01-10:
I'm going to close this as I think the original question was answered.

-Kevin

closed 2019-01-10 by KevinOConnor