Klipper3d/klipper #1060: Thoughts on a Klipper Slack Team

dotps1, 2019-01-01:
I have just noticed that a lot of 'issues' are questions, and there seems to be enough users here I think we could all benefit from a slack team.  Just wondering everyone's thoughts on that.

jakep82, 2019-01-01:
There is already an IRC channel, a mailing list, and an unofficial Discord group.  I don't think we need another method for communicating.

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Scottbg1, 2019-01-01:
Considering I am not on IRC, or Discord and I get enough email notifications each day my phone sounds like a winning slot machine, it would be nice if something like a slack team were implemented.

I looked for a forum where Klipper might have been a main subject but I found none.
Even if I I could get my answer from someone like me who might have been surfing though here and realized that they had the same issue and could give me an answer it would take a load off the developer(s) and concentrate on serious bugs and issues.

Just my $.02

theopensourcerer, 2019-01-01:
There's also a fairly active unofficial group on Facebook too with nearly 3000 members.

https://www.facebook.com/groups/klipper/

tripstar76, 2019-01-01:
There's also https://www.reddit.com/r/klippers/

dotps1, 2019-01-02:
So, I am in a very similar situation at @Scottbg1, and im not on IRC or Discord, and defiantly get enough email notifications lol.  And I really like my other slack teams for specific topic questions and generally very fast responses.

That being said, I created a team, klipper.slack.com.  I am more then willing to transfer ownership to whomever it should go to.

So, if you'd like to have an invite sent you can reach out here, or contact me (contact email is in my profile).  And I will shoot you an invite.

I have a nice way to use google forms for an automated invite system if the klipper devs want me to set that up, I will.  But I am not in anyway part of the development of this project, so, not sure if that something I should do.

thanks!

glowtape, 2019-01-02:
If any Chromium based messaging service, I'd sure prefer Discord over Slack. Former performs faster and doesn't expect me to pay a subscription for scrolling up too far back or leaving my private message history accessible.

KevinOConnor, 2019-01-03:
FYI, this has come up a few times before - variously with suggestions for forums, wikis, facebook, slack, discord, etc.  I agree the current system (github, mailing list, irc) could be improved.  I'm open to moving one of the official communication channels to something else.  The mailing list hasn't really been utilized at all and the IRC channel isn't super active.  FWIW, I don't think facebook is in keeping with the open source nature of this project and I'd be concerned about slack as well for that reason.  An official Discord server does seem reasonable to me.  I'm open to options and open to trying an alternative.

-Kevin

dotps1, 2019-01-05:
I think my only preference for slack is I already use it for many other teams, and I don't really want MS Teams, Slack, Discord all running on just for chat lol.  I know the powershell slack team I am on has a discord bridge, that connects the two, or atleast allows you to connect to the slack conversation, on one channel from discord.  Not sure if something like that is an option or not.

KevinOConnor, 2019-02-18:
I'm going to close this as it looks like the conversation has concluded.  I'll continue to look at alternatives for improving communication.  (I agree that the project could improve here.)

-Kevin

closed 2019-02-18 by KevinOConnor