Klipper3d/klipper #1061: Z Baby Stepping

Scottbg1, 2019-01-01:
I have tried to find a definitive answer to my question, but I haven't come across it, so please forgive me if this has already been answered...

I found documentation explaining  the SET_GCODE_OFFSET command and how it effects an axis offset.
I was of the assumption that it could be used as Z Baby Stepping and it does to a degree.
But what I found is it only allows one change during printing.

Meaning, I created Macro buttons, one with SET_GCODE_OFFSET Z=.05 for +baby steps, and another macro with SET_GCODE_OFFSET Z=-.05 for -baby steps. But it seems during printing it only allows for one instance of a button press, or one change during the duration of a print.

I am new to Klipper and pretty new to Octoprint. I have another printer with a board that has a built in web server that I have used for quite awhile and this type of baby step function is standard with it.
I was trying to mimic this with buttons on Octoprint as it is quite useful.

Am I correct in my assumption that by design Klipper only allows for one change during the print, or am I missing something

Scottbg1, 2019-01-01:
I think my question has already been asked...

https://github.com/KevinOConnor/klipper/issues/1037

I will wait to see what becomes of this conversation.

timryder, 2019-01-01:
Klipper will Que the command to change the Z height and it will execute the change when it is reached in the que.  There is discussion currently to allow for the  SET_GCODE_OFFSET Z to be a Real-Time command which does not observe the cue but it is handled instantly. There is the discussion at #1037 as you mentioned yes.

Scottbg1, 2019-01-01:
Thanks Tim.

One thing...
Since we know the command is sent to que before implementing, am I wrong in thinking that it only executes once per print? Or will it indeed allow adjustment multiple times until desired height is achieved?

KevinOConnor, 2019-01-03:
It sounds like you want to use `SET_GCODE_OFFSET Z_ADJUST=0.05`, otherwise each time you run the command it will reset the distance to the absolute offset of .05.  See the description at: https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#extended-g-code-commands

-Kevin

Scottbg1, 2019-01-03:
Thanks for taking the time to answer Kevin.

I read the description you suggested and I took this from it...

This is how I have my setup now:
If my z height at the beginning of a print were zero, and during my first layer I noticed the z height was too low, the macro button I created to raise the nozzle would send the command `SET_GCODE_OFFSET Z=.05` to do just that, by .05mm. If again I noticed it were too low, I hit my macro button again to raise the nozzle yet another .05mm I would have then changed the z height by .1mm higher. Or at least I hope it would.

If I understood your suggestion and the description I just read, I should instead be sending `SET_GCODE_OFFSET Z_ADJUST=.05` instead to raise the nozzle each time I need it...Correct?

If so, part of my answer is satisfied, the other being WHEN that command takes effect, per the other conversations going on at the moment about the subject.

I'll change the commands on my Macro buttons and give it try!

KevinOConnor, 2019-01-03:
Yes - you want the Z_ADJUST version.  It takes effect on all g-code commands after the given SET_GCODE_OFFSET command.

-Kevin

timryder, 2019-01-04:
Kevin - 
Are the Extended G-Code commands volatile? Meaning they will no longer be active after a power cycle? Or do they get written to some form of memory where they're always active? 

Thanks

KevinOConnor, 2019-01-06:
The only g-code command that would alter state between restarts is the SAVE_CONFIG command.

-Kevin

KevinOConnor, 2019-01-16:
I'm going to close this as I think the original question has been answered.

-Kevin

closed 2019-01-16 by KevinOConnor