Klipper3d/klipper #1063: flashing problem

Fabrizio3, 2019-01-02:
im flashing my a8, it does flash into the memory but then it doesnt connect, i checked the klippy logs and its trying to connect to ttyACM0 instead of ttyUSB0, i have specified USB0 in the config process, but where do i change it so it actually uses it?
[klippy.log](https://github.com/KevinOConnor/klipper/files/2719817/klippy.log)

LemarinelNet, 2019-01-02:
Hi,

In the config process, you specify which UART port of the AVR processor should be used. It is not related to the usb port that is used by klippy.

The USB port used by klippy is set in the printer.cfg. Have a look in this file ;)

dotps1, 2019-01-02:
with my ender I always have to do it by ID in the config file under `[mcu]` section:

https://github.com/KevinOConnor/klipper/blob/e4e0fec681a934adc7592e080e2653e0e808fdec/config/printer-anet-a8-2017.cfg#L71

KevinOConnor, 2019-01-06:
I recommend following the steps at: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

-Kevin

Fabrizio3, 2019-01-06:
the problem turned out to be a bug, had set usb0 as default in the printer.cfg, changed it to something else then back to usb0 and it started working
thanks anyways! klipper is an amazing firmware, and being able to push 300mm/s at 2200 acceleration and still getting better quality than marlin is plan amazing!

KevinOConnor, 2019-01-16:
Thanks.  I'm going to close this as it looks like the issue was resolved.

-Kevin

closed 2019-01-16 by KevinOConnor