Klipper3d/klipper #1065: Controlling several fans

lmcbmai, 2019-01-02:
Hi, 
This is my 1st post here and I'd like to start with saying THANKS to Kevin for outstanding job done on Klipper project. It was just 2 days ago when I heard about Klipper the very 1st time and I already have my printer  up and running with Klipper and OctoPrint. It is extremely rear when you find such a good documentation and clear installation guidelines. 

Almost 3 years ago I purchased my first 3d printer - Velleman's Vertex K8400 and have been modifying it since. All my mods are published on Thingiverse and I can share the links, if someone is interested. Fast forward to present day - my printer still has the original mainboard, LCD and SD card reader from Velleman. Kinematics was converted into H-bot. Hotends were replaced with two e3d-v6 sharing the same fan. Hotbed was added. Until yesterday it was running Marlin with OctoPrint. 
I installed Klipper, configured all pins and even printed the 1st test cube - all in a single day! LCD is also working. 

Now to my questions. They are about controlling the fans. I'd like to know what would be the best configuration for the following 4-fan setup:

The first fan is cooling down both e3d hotends, i.e. if the temperature of ANY of them goes above 80C, I'd like the fan to be powered on.
- 2nd fan is blowing on the print. I configured it under [fan] section of printer.cfg.
- 3rd fan is cooling down the mainboard. In the original K8400 it was powered on any movement along X,Y or Z axis and powered off 1 minute after the last move. Right now I have this fan controlled by [static_digital_output my_output_pins], i.e. it is always on. 
- I'd like to have addition fan blowing on Raspberry Pi, that should switch on when CPU temperature rises above certain level. 

My mainboard has only 2 outputs for fan control. I have extension board for Raspberry that can control 3 additional fans. So, which of the above 4 fans would you recommend to control from the mainboard and which from Raspberry? 

Thanks
Best regards,
Boris

KevinOConnor, 2019-01-02:
Thanks.

> The first fan is cooling down both e3d hotends, i.e. if the temperature of ANY of them goes above 80C, I'd like the fan to be powered on.

I'd use a "[heater_fan]" config section (see config/example-extras.cfg).  The "heater" parameter can take a comma separated list of heaters - so just list both of your extruders there.

> 2nd fan is blowing on the print. I configured it under [fan] section of printer.cfg.

Right - use a "[fan]" config section.

> 3rd fan is cooling down the mainboard. In the original K8400 it was powered on any movement along X,Y or Z axis and powered off 1 minute after the last move. Right now I have this fan controlled by [static_digital_output my_output_pins], i.e. it is always on.

This has come up a few times before.  Possible solutions are to use a heater_fan and just have it turn on when one of your extruders are on, use an output_pin and manually turn it on/off via g-code commands, use a multi_pin and attach it to one of your stepper enable lines so that it turns on/off when one of your steppers turn on/off, or to add a thermistor near your board and configure a [temperature_fan] config section so that it turns on when the board heats up.

> I'd like to have addition fan blowing on Raspberry Pi, that should switch on when CPU temperature rises above certain level.

That's interesting.  Simplest would be to add a thermistor and use a temperature_fan.  I'm sure there are also other solutions, but it's not something I've played with.

-Kevin

KevinOConnor, 2019-01-02:
Oh, to be clear, there isn't currently support for controlling fans directly connected to the raspberry pi.  It's something that's possible to implement, but it would require software development.

-Kevin

lmcbmai, 2019-01-02:
Thanks for quick response. I'll try your suggestions tonight. 

As for measuring temperature of Raspberry CPU - there is no need in thermistor. Here is the command that shows the temperature followed by a sample output
```
pi@raspberrypi:~ $ /opt/vc/bin/vcgencmd measure_temp
temp=56.9'C
```
I will have fan connected as described in this video https://www.youtube.com/watch?v=oJ32CMxliCQ

Thanks.
-b

Master92, 2019-01-02:
>> 3rd fan is cooling down the mainboard. In the original K8400 it was powered on any movement along X,Y or Z axis and powered off 1 minute after the last move. Right now I have this fan controlled by [static_digital_output my_output_pins], i.e. it is always on.

> This has come up a few times before. Possible solutions are to use a heater_fan and just have it turn on when one of your extruders are on, use an output_pin and manually turn it on/off via g-code [...]

Actually, I am working on a solution for that case right now. I am planning to implement a fan that turns on at a specified speed as soon as a stepper or specified heater(s) are active, changes to an idle speed, as soon as these are being deactivated, and turns off after a configurable timeout.

I actually already have the code written by the time writing and will be issuing a pull request by tomorrow or the day after that, if Kevin states here, that this is a good feature for Klipper.

-Nils

swatti87, 2019-01-06:
Add "Motor_fan" maybe?

Motors on, said fan turns on. Usefull for cooling drivers and motors.

Master92, 2019-01-06:
>Motors on, said fan turns on. Usefull for cooling drivers and motors.

That's almost exactly what my module from #1070 does, with the addition that the said fan also turns on, whenever the extruder and/or another heater is activated.

-Nils

KevinOConnor, 2019-01-16:
I'm going to close this as I think the original question was answered.

-Kevin

closed 2019-01-16 by KevinOConnor

goosst, 2020-11-29:
@lmcbmai : would you mind sharing your configuration file for the k8400 please?

lmcbmai, 2020-11-29:
Hi @goosst,
Couple of months after starting this thread I designed and built another 3d printer with 32bit mainboard that uses Repetier firmware. For ease of maintenance I migrated K8400 to Reperier, and have not been using Klipper since. 
I may have configuration backups somewhere on my home hard drive, but I am out of country till at least end of January. Another thing to consider - my K8400 has dozens of mods, and by the time I run Klipper on it printer has already been converted into H-bot with heated bed. That's why there is a good chance that my firmware configuration won't work for you.
In any case, if you will be still interested, ping me again in early February and I will search my backups for you.

Thanks

moebis, 2021-09-02:
@KevinOConnor Is it possible to set a condition to turn on the controller fans when the Raspberry Pi reaches a certain temp? On my Voron those 2 fans basically cool all of the electronics, and turn on when the Bed Heater reaches 50 degrees, but there are times that I also want to turn them on when the Pi or MCU reaches a certain temperature. I tried something like this to no avail in the printer.cfg:

```
[fan controller_fan]
##    Controller fan - pi Connector
pin: PB2
fan_speed: 1.0
shutdown_speed: 1.0
kick_start_time: 0.5
temperature_sensor: raspberry_pi
temperature_sensor_temp: 50.0
```

angeljrgsr, 2021-10-31:
> I'm going to close this as I think the original question was answered.
> 
> -Kevin

@KevinOConnor we just need to have the raspberry pi's sensor accessible, much like a heater, as @moebis states above.

KevinOConnor, 2021-11-01:
Not many people will see your update to this old issue ticket.  Best to ask via the directions at https://www.klipper3d.org/Contact.html .

-Kevin

angeljrgsr, 2021-11-01:
> Not many people will see your update to this old issue ticket. Best to ask via the directions at https://www.klipper3d.org/Contact.html .
> 
> -Kevin

Thanks! It looks like this is implemented but not well documented. I was able to get help on Discord.