Klipper3d/klipper #1066: Installation Issue

ShaunCro, 2019-01-02:
Hi Kevin,

Been dying to try out your firmware but run into an issue after installation that no amount of googling and re-installing has helped with.

Printer details:
Modified Wanhao Duplicator 4X
RAMPS 1.4 controller board with Arduino Mega 2560
A4988 Drivers
E3D V6 Single Hotend
100K Ohm, NTC, 3950 1% on the Bed and Hotend.

Error I get after installing and trying to connect and issue commands to the ramps.

Error: Unable to parse option 'max_z_accel' in section 'printer'
Unable to parse option 'max_z_accel' in section 'printer'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready

Working fine with Marlin 1.1.9

Klippy.log attached. Any and all help is greatly appreciated.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2721451/klippy.log)

theopensourcerer, 2019-01-02:
Remove the line 

> 	"RepRapDiscount 2004 Smart Controller" type displays

Or comment it out :-)

ShaunCro, 2019-01-02:
Is that all O.o 

Sigh, you can tell I'm not a coder. Thanks, will give it a go when she's finished her current print and let you know :-)

theopensourcerer, 2019-01-02:
If it works - just close this issue.

ShaunCro, 2019-01-02:
Will do thanks. Have 60 mins to go on the current print job and then will swop out the Mega and test. Thanks for the assist dude.

ShaunCro, 2019-01-02:
Dude, you are my hero. She works! My FrankenPrint lives. 

Thanks man. Now to get down to the configuration.

closed 2019-01-02 by ShaunCro