Klipper3d/klipper #1068: Multi MCU

nandinho84, 2019-01-02:
Sorry, this may be a silly question for some of you, but I can't seem to find the answer anywhere.
I'm trying to run klipper with two MCU (one is arduino - fysetc F6 and the other is the MKS Sbase 32 bits). I was reading the "example-multi-mcu.cfg" and it states "pin_map: arduino". I was wondering what am I supposed to put on the pin_map for the MKS Sbase (the chip is the lpc1768) - or where can I get this information? Please help. Thanks!

trevjonez, 2019-01-02:
if I understand correct, the pin map is just a convenience thing to make dealing with ramps or the F6 easier since the pinouts there are typically the D56 or w/e style labels whereas the atmega chip is actual like PB0-PB7, PC0 etc. so for the MKS it would probably be something closer to what is used in the generic-smoothieboard.cfg which seems to match what is on the images here: http://3daddict.com/mks-sbase-32bit-mainboard-controller/

and f6 from here: https://wiki.fysetc.com/F6_V1.3/#hardware-reasources shows the D pin numbering style.

nandinho84, 2019-01-02:
Thanks for the reply. 
I know that the MKS Sbase uses the pins like the smoothieboard (i.e. P2.4; P0.25, etc) however, what I don't know is what to fill in in the "pin_map" option... Can I ignore this line? Or do I have to put something like "pin_map: smoothieboard" or "pin_map: lpc1768"???

trevjonez, 2019-01-02:
looks like you don't need that line for it to work. I imagine it will throw an error if your pins don't match for the MCU that is connected.

jakep82, 2019-01-03:
> looks like you don't need that line for it to work. I imagine it will throw an error if your pins don't match for the MCU that is connected.

This is correct.  Take a look at the generic-smoothieboard.cfg file for reference.  It doesn't call out a pin map.

nandinho84, 2019-01-03:
Ok, thanks a lot guys!

closed 2019-01-03 by nandinho84