Klipper3d/klipper #1069: first heating bed, then extruder nozzle.

peterlpop, 2019-01-03:
Is it normal for the heaters to go one at a time - stock firmware heated them both (and I "think" Klipper did it as well on the first day I had it installed. Did I change something and I can't remember? Thanks.

Master92, 2019-01-03:
It depends on what commands you send to the printer. For example when you issue an M140, the printer heats up the extruder to the specified temperature and waits for the process to be completed, before executing the next command. However an M104 does the same thing without waiting. Same for M109 and M190 for the heated bed.

KevinOConnor, 2019-01-10:
I'm going to close this as I think the original question was answered.

-Kevin

closed 2019-01-10 by KevinOConnor