Klipper3d/klipper #1071: "pin virtual_endstop used multiple times in config" is desired

towe96, 2019-01-03:
I'm trying to set up my CoreXY printer using TMC2130s on a MKS Gen 1.4 board.
Since it's CoreXY, I only need one drivers diagnostics pin for sensorless homing.
In Marlin, I simply set the Y endstop to use the same pin as the X endstop, and connected the XMIN sensor pin (ar3) to my Y driver.

I attempted to do the same in the Klipper config file, by using the "tmc2130_stepper_y:virtual_endstop" on both "stepper_x" and "stepper_y"- this resulted in an error at startup (restart -> status) since the endstop was used multiple times. The log is attached here:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2723771/klippy.log)

What would be the correct procedure / config for sensorless homing on a CoreXY printer?

towe96, 2019-01-03:
Homing seperately is required for sensorless homing anyways, even in Marlin (e.g., not using "QUICKHOME").
I don't see why I wouldn't be able to use the virtual endstop for both axes though.
I assume it would be fine to use two virtual endstops (X and Y), but have them poll the same pin (ar3, connected to the Y driver)?

jakep82, 2019-01-03:
You can't use the same pin twice for anything in Klipper. You need to configure separate X and Y pins, even on a corexy.

towe96, 2019-01-03:
I noticed that too now - that's just.....

I feel like Klipper really needs CoreXY support in that regard. Multiple pins are just not required on CoreXY for sensorless homing.

jakep82, 2019-01-03:
Sensorless homing is being discussed [https://github.com/KevinOConnor/klipper/pull/1004](url) here.  For now it requires separate endstop pins.

KevinOConnor, 2019-01-03:
Support for detecting stalls on just one stepper during both X and Y homing would require software development.

If you're comfortable making code changes, you could try adding a call to `self.printer.lookup_object("pins").reset_pin_sharing(pin_params)` at the start of the setup_pin() definition in klippy/extras/tmc2130.py.  I don't know if this will work or what odd effects it may lead to.  If you make a code change, be sure to run "sudo service klipper restart" after each change.

If you're not comfortable making a code change then this would need to wait for a developer interested in working on it.

-Kevin

towe96, 2019-01-03:
I've build an adaptor for now, connecting the Y diag pin to two input pins on the MCU, which are then set up as (virtual) end stop pins in my config.

However, regardless of driver_SGT settings, it does not seem to react to running into the carriage - I end up having to manually reset the printer to stop it from jerking around.

In Marlin, a threshold of 2 worked just fine.

jakep82, 2019-01-03:
I don't think that will work.  Klipper sets the steppers to a specific mode when homing, so when homing X, only the X stepper will send the proper signal via DIAG1.  At least that's what I believe is happening based on what I see in tmc2130.py, and my very basic understanding of how sensorless homing works.  So at least for now, you need to run a wire from both your X and Y steppers.

towe96, 2019-01-03:
You're right.
I've soldered on a header to the X driver as well now (which I had wanted to avoid) and set up the endstops as usual - it homes fine now.

I guess Marlin does handle that differently though, since apparently both drivers are in "homing" mode - just mapping the Y pin to the X pin was sufficient for that.
It's odd though, since both drivers should be in "homing" mode on a CoreXY system to begin with in my opinion. It also had the (maybe?) added benefit of having just one driver with just one StallGuard threshold doing all the homing work.
I think the only change in the code would be to use the same procedure for X and Y homing when coreXY is used with sensorless homing - basically only the travel direction would have to be changed.

yhaiovyi, 2019-01-05:
Okey, another use case is same tmc2208  uart pin for double motors.

KevinOConnor, 2019-01-16:
I'm going to close this as I think the original question was answered (Klipper does not currently support sensorless homing in the original poster's configuration).

-Kevin

closed 2019-01-16 by KevinOConnor