Klipper3d/klipper #1072: probe Z height

athlon789789, 2019-01-04:
Hi I am new to klipper 
my setup is a DIY delta printer with a proximity sensor probe.The klipper is working pretty fast and smooth on my machine compare to 8bit marlin.I am only using delta calibrate but no ABL.
I wonder to know if it is possible to probe a Z height and make a G-code offset at beginning of every print
beacuse my printer is relatively bigger in Z height（400MM）and the actual Z height will slight change under different room temperature.
sometimes it will drifting 0.3MM or more that will significant impact the frist layer bonding.
In marlin I am adding a "G33 P1"in the start of Gcode file,the printer will do a Z height probe and make a change in Z height automatically .So it will greatly increase the success rate of prints.
wish you all happy New Year!

KevinOConnor, 2019-01-04:
It's typical to use G92 for this purpose - something like: `G28, G1 Z10, PROBE, G92 Z<offset>`, where <offset> is the probe z offset.

-Kevin

athlon789789, 2019-01-05:
thanks Kevin I will give it a try

KevinOConnor, 2019-01-16:
I'm going to close this as I think the original question was answered.

-Kevin

closed 2019-01-16 by KevinOConnor