Klipper3d/klipper #1074: unknown GCODE "?$G"

troisplus, 2019-01-04:
[klippy.4.1.19.log](https://github.com/KevinOConnor/klipper/files/2727167/klippy.4.1.19.log)
[image]
[image]

After several tests and a satisfactory first print, I want to solve a problem appeared from the beginning. Octoptint keeps sending an unknown GCODE (?$G to the terminal). I think this problem is related to the absence of a temperature check on the graph (yet functional during printing or PID). The command M105 does not answer either. Do you have an idea?

john--, 2019-01-04:
Hi @troisplus , are you trying a klipper stable release, e.g. 0.7 or did you check out from master? I ask because master got quite a few merges recently, including something I personally put in around heater reporting. Not sure if that's the issue but I'd be curious to see if it's a new issue or not.

john--, 2019-01-04:
Also, just one other thing to consider, is it Klipper or Octoprint at fault? A quick google search showed someone else seeing the same behaviour as you after installing a buggy CNC related Octoprint plugin. Not saying it's that, but something to keep in mind.

troisplus, 2019-01-05:
OMG, i'm sorry...
I was however convinced to have disabled all the plugins off octoklipper and preheat ...
This is effectively the GRBL plugin fault.

KevinOConnor, 2019-01-16:
I'm going to close this as it looks like the issue was resolved.

-Kevin

closed 2019-01-16 by KevinOConnor