Klipper3d/klipper #1079: Setting up BLTouch and Mesh Bed Leveling How To for Newbies

dotps1, 2019-01-05:
I apologize in advance for this question, and i'm sure all the information is in the docs, but I am just overwhelmed with all the information and configuration options.  I only used marlin for about 5 days in my life of 3D printing before I moved to klipper and have been here pretty much ever since (which is only about 3-4 months in total).  That being said, I got a BLTouch sensor for xmas, I printed the mount and I have it installed.  I fumbled my way through trial and error and google enough to at least get it to work as the Z end stop, so my hot end doesn't crash into the bed.  Following other configs people have shared online, basically when I start a print, it goes to the center of the bed, probes, then moves to X0 Y0 then starts printing.  Shouldn't it be doing the 5x5 mesh bed probing before every print (which I think is in my config, config is attached)?  Do I define that in the homing_override section?

Any way to my question, is there any good walk through or how to for setting up bltouch and meshbed leveling for a _beginner_ using klipper out there?  Im not super familiar with the GCodes yet so I really have no idea what i'm doing, especially with the homing_override.  But at this point, unless I put the  orginal  Z end stop back on, I cannot print anything.

I was also reading the bed_mesh_calibrate info, and do I run that before every print, in my homing override?  I'm just confused.

Like I said, im sure the info on what I need is all in the docs, im just having a hard time getting it all put together and what settings apply to me and my Ender 3.

Thanks for any help!
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2729768/printer.cfg.txt)

BlackStump, 2019-01-05:
these are the macro's I use with my Ender 3, Bed Mesh levelling could be enabled to probe every time before a print but that would be a waste of time imho because you can save the mesh and load that mesh before a print as seen in my macro.
The important bed mesh commands are here and entered in the terminal.
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#mesh-bed-leveling

[gcode_macro G10]
gcode:
    G91
    G1 E-2.5 F4000
    G90

[gcode_macro G11]
gcode:
    G91
    G1 E2.5 F1200
    G90

[gcode_macro start_g]
gcode =
  G28
  BED_MESH_PROFILE LOAD=ender3
  G1 F8000
  G1 X234 Y-0.4 Z5 F3000 
  G1 E15 F300           
  G92 E0                
  G0 Z0.5              
  G1 X160 Y1 E10 F200      
  G92 E0               
  G1 X120 Y1 F3000
  
[gcode_macro end_g]
gcode =
  G10
  G91
  G1 Z+10
  G1 E-10 F300
  G90
  G1 X10 Y220 F2000
  M104 S0 
  M140 S0
  M84 
  BED_MESH_CLEAR 

[bed_mesh]
samples: 3
sample_retract_dist: 8
min_point: 50,3
max_point: 235,220
probe_count: 5,5

dotps1, 2019-01-05:
thanks for the reply @BlackStump can I ask a few more follow up questions:
1. did you replace the springs on the bed mounts with something static so the bed "can't" move?  or do you just re do the mesh leveling terminal commands every so often?

2. am I following this right, you basically define the bed mesh in the config, then when you use the calibrate commands, it runs off that, then you save that output to profile name, then that is what is loaded in with the `BED_MESH_PROFILE LOAD=ender3` gcode line?

thanks again, when I get some time later this evening I will try and incorporate some this in my config.

BlackStump, 2019-01-05:
I still have the springs as original, I do not have to redo the mesh often at all and have consistent first layers.
You need to manually run BED_MESH_CALIBRATE then when that completes you use the command
BED_MESH_PROFILE SAVE=ender3
then 
SAVE_CONFIG as per the message you will receive in the terminal.
you can then load that ender3 mesh before each print as I have done in my slicer start_g

dotps1, 2019-01-05:
ok ok, this all starting to make sense to me.  Im guessing I need to set up my z offset first right?  is there a good way to go about calculating that?

BlackStump, 2019-01-05:
another user has made this doc as a PR so might be helpful for you
https://github.com/KevinOConnor/klipper/blob/ebede9827c6a8b8e194b213e143e7a72da4fec36/docs/BLTouch.md

dotps1, 2019-01-05:
ok, I did those steps and now have my x,y,z offsets, do you use a homing override?  also I see you define a G10 and G11 macro, but I don't see the G10 macro called in your start stop gcode, can you help me under stand that part?

BlackStump, 2019-01-05:
yes I use a homing override the G10/G11 macro is firmware retraction instead of using a slicer retraction not every slicer supports that so it is a option to be used or not, G10 is retract and is in the end_g macro and G11 is unretract.
The thing with macros you can create one for whatever purpose you are trying to achieve.
The start_g is a purge routine for my Ender3 so the extruder is primed and ready to print.
[homing_override]
axes: z
set_position_z: 5
gcode:
G90
G1 Z10 F600
G28 X0 Y0
G1 X160 Y120 F3200
G28 Z0
G1 X0 Y0 F7200

BlackStump, 2019-01-05:
gcode is something that needs to be referenced and this is the site to use for that 
https://reprap.org/wiki/G-code

dotps1, 2019-01-05:
awesome!  thanks for all the help, I have the offsets, and the mesh bed calibration done, I was even able to get it working in the bed visualizer plugin.

I printed a right side mount and I think that is no good for the ender, because it makes the first probing already about 50mm in on the X, so I have your startup gcode in, im going to try printing a left side right now.

thanks again for the help @BlackStump I feel like i'm making progress!

BlackStump, 2019-01-05:
yeah the left side makes more sense for the Bltouch ...great to hear you are making inroads

dotps1, 2019-01-05:
Its printing, and the first layer stuck amazing, I think the only change I want to make is have it pick up the z a little when it does its first move after purging the extruder.  thanks again!

closed 2019-01-05 by dotps1

ManuInNZ, 2019-03-10:
@dotps1 Any hint to get the bed visualiser plugin to work nicely ?

NicoSoftware, 2019-05-26:
@BlackStump 
> The macros start_g and end_g are used with slicers to make it consistent with all slicers I use

How do you call the macros from slicers to load the profile?

BlackStump, 2019-05-26:
most slicers have a start gcode/script and a end gcode/script section so you add it there ie S3D example

[image]

NicoSoftware, 2019-05-28:
> most slicers have a start gcode/script and a end gcode/script section so you add it there ie S3D example
> 
> [image]

Cool thing, will look into that with Cura.
Thanks alot :)

NicoSoftware, 2019-05-30:
> most slicers have a start gcode/script and a end gcode/script section so you add it there ie S3D example

I've tried it with cura either way calling the script name or a GCODE Macro with the ```BED_MESH_PROFILE LOAD=ender3``` stops my print instantly.
even tho it's saved in the config..

BlackStump, 2019-05-30:
post your klippy.log file maybe your macro is configured wrong

NicoSoftware, 2019-06-03:
Hey, have been busy the last days, here it is!

Klippy.log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3249169/klippy.log)
this is my printer config:
[printer.txt](https://github.com/KevinOConnor/klipper/files/3249173/printer.txt)

BlackStump, 2019-06-03:
delete your G10 G11 macros that call start_g and end_g, G10 G11 is normally reserved for firmware retraction
ie

[bltouch]
sensor_pin: ^PC4 ; P1.24
control_pin: PA4 ; P1.26
pin_up_reports_not_triggered: true
pin_up_touch_mode_reports_triggered: False
pin_move_time: 0.200
x_offset: -46
y_offset: -2
z_offset: 0.002500  # that does not look right 1.55 is closer to the normal offset BLtouch
speed: 5

[gcode_macro start_g]
gcode =
    G28
    BED_MESH_PROFILE LOAD=ender3
    G1 F8000
    G1 X234 Y-0.4 Z5 F3000
    G1 E15 F300
    G92 E0
    G0 Z0.5
    G1 X160 Y1 E10 F200
    G92 E0
    G1 X120 Y1 F3000

[gcode_macro end_g]
gcode =
    G91
    G1 Z+10
    G1 E-10 F300
    G90
    G1 X10 Y220 F2000
    M104 S0
    M140 S0
    M84
    BED_MESH_CLEAR

NicoSoftware, 2019-06-04:
Hey,
cleared G11 and G10, I just used it because the start_g didn't worked with Cura.
Now my printer.cfg is looking like this:
[Uploading fz3-15596399035060…]()
But same procedure, he auto homes within the script and stops immediatly after..
Klippy.log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3251535/klippy.log)
In the terminal it says "unknown command M108":
[image]
And in Cura it looks like this:
[image]

BlackStump, 2019-06-04:
I think you need to do some research on G-code
https://reprap.org/wiki/G-code
and the G_code the Klipper supports
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md
and how to use klipper macros
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L1471
Then look at Octoprint defaults that may need changing to better support Klipper because Octoprint is not Klipper.
May you should delete the the start_g and end_g macros until you understand what they are actually doing. They are not required to run klipper with any slicer.

NicoSoftware, 2019-06-04:
So it's working now, I've basically just cut out all Gcodes, which has been written after the profile load.
I know GCode and I really thought that everything was okay with the script.
Anyway, it works now.
He's constantly changing the x coordinates, thank you alot!

angrymrtom, 2019-06-04:
I thought I'd drop this here, I am the one who opened this issue as @dotps1, anyway, after my experience with using BLTouch, Klipper, and the Ender3, I tried to collate all my knowledge, this is what I set up, and I how I calibrated it, and how I continue to calibrate it, especially after I replace the nozzle.

https://angrymrtom.github.io/using-bltouch-with-klipper-and-they-ender3/

if it helps someone great!

jakep82, 2019-06-04:
> You need to allow the Z Axis to go below the ‘0’ mark for negative warp, this can be done by editing the endstop_min value, also in the z_stepper section.
endstop_min: -3

That should be position_min.

angrymrtom, 2019-06-04:
updated, thanks.

KevinOConnor, 2019-06-05:
Thanks.  FYI, there's been some new documents in Klipper - the bltouch document ( https://www.klipper3d.org/BLTouch.html ) and a new command to assist in finding the z_offset ( https://www.klipper3d.org/Probe_Calibrate.html#calibrating-probe-z-offset ).  There's also some draft documentation currently in the works ( https://github.com/KevinOConnor/klipper/blob/work-probe-20190601/docs/Probe_Calibrate.md ).

-Kevin