Klipper3d/klipper #1081: M112 Fails to shut down during M190 bed temperature heating

archer183, 2019-01-06:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2730216/klippy.log)

M112 does not function when M190 is commanded by a print until the heatup cycle is complete.   M112 DOES interrupt the heatup cycle if M190 command issued through octoprint terminal rather than Gcode.

Log file shows both tests on 1/5/2019.   First test was M190 issued from terminal, where M112 does interrupt.  Second test was executing a print job and attempting to interrupt with M112 from octoprint terminal, no response to M112 until after heatup completed.

john--, 2019-01-06:
Hi @archer183 , which version are you using 0.7? or a checkout from master?
I ask because I made a change to temperature reporting recently and that change is in master now (but not in a version release). It's unclear to me if this is related to my change or something pre-existing.

I can reproduce this, but from what I can see the MCU and the heaters are properly shutdown, only klippy  (via M105) still report that there is a set temperature until you perform your FIRMWARE_RESTART. I think you're seeing similar behaviour since I can see your temp dropping even those the target is showing 100.

From what I can see a "RESTART" command still properly turns off all heaters like it used to, so it doesn't seem like I introduced a breaking change.

@KevinOConnor  any insight here? Anything I could potentially help fix?

archer183, 2019-01-06:
I was able to verify the heater was still receiving power after m112 was
issued as I drive my heater with a scr that lights when active, and the
temperature kept going up after m112.    I installed early December the
normal way

john--, 2019-01-06:
Thanks for the reply.
Interesting, because your log originally shows:
`heater_bed: target=100 temp=99.2`
but by the end it reports
`heater_bed: target=100 temp=80.5`
This suggests to me that the heater was physically off, but I could be completely wrong.
Maybe it would be best for someone more knowledgable to chime in here. This only caught my attention because it sounded related to something I'd be working on.

archer183, 2019-01-06:
Let me get a clean log tonight assuming it replicates again, as I did some
terminal tests that did work but when running a print job it did not in
that log

archer183, 2019-01-06:
So I just replicated the issue, however it does not appear to be a klipper issue.  it is an octoprint issue.  Octoprint appears to be ignoring terminal commands during the heatup (including M112 command ) but will send the cancel command from the main window if you cancel the print.   Log attached just in case I am wrong in my diagnosis 
[klippy.log](https://github.com/KevinOConnor/klipper/files/2730863/klippy.log)

KevinOConnor, 2019-01-07:
It looks like Klipper did the right thing here.  If you run your logs through the logextract.py program (see docs/Debugging.md for the details), you'll see for the three shutdown events:

```
Read 303.438563: 'M112\n'
Sent 99 303.441730 303.441490 6: seq: 18(456), emergency_stop
Sent 99 303.441864 303.441624 6: seq: 15(693), emergency_stop
Receive: 99 303.444309 303.441730 12: seq: 19(456), shutdown clock=428162181(303.441723) static_string_id=8
Receive: 99 303.449049 303.441864 12: seq: 16(693), shutdown clock=415132473(303.441349) static_string_id=8
```

```
Read 975.043884: 'M112\n'
Sent 99 975.046589 975.046349 6: seq: 1b(859), emergency_stop
Sent 99 975.047168 975.046568 15: seq: 1a(2186), emergency_stop, schedule_soft_pwm_out oid=7 clock=1755095374(975.308984) value=108
Receive: 99 975.051867 975.046589 12: seq: 1c(859), shutdown clock=1767716428(975.046759) static_string_id=8
Receive: 99 975.054344 975.047168 12: seq: 1b(2186), shutdown clock=1750912028(975.047750) static_string_id=8
```

```
Read 362.625455: 'M112\n'
Sent 99 362.627124 362.626884 6: seq: 1f(559), emergency_stop
Sent 99 362.627261 362.627021 6: seq: 15(1189), emergency_stop
Receive: 99 362.632103 362.627261 12: seq: 16(1189), shutdown clock=1362028124(362.627234) static_string_id=8
Receive: 99 362.639346 362.627124 12: seq: 10(559), shutdown clock=1382106086(362.626811) static_string_id=8
```

In all three cases, Klipper requests the micro-controller to "emergency_stop" and gets an acknowledgement back from both micro-controllers all within a few milliseconds of the time Klipper reads the M112 request.  If OctoPrint did not send the M112 to Klipper there's not much Klipper can do.

When you typed M112 into the OctoPrint terminal, did OctoPrint immediately disconnect from Klipper?

-Kevin

archer183, 2019-01-07:
No.   It waits until the heating is done.     That is why I think it is
octoprint failing to send the command in the first place

Arksine, 2019-01-07:
Yes, it is definitely an Octoprint issue.  When printing from Octoprint it will wait to send M112 until heating is complete.  However, you can use Octoprint to print from SD.  When it isn't in processing mode, it will allow M112 to pass and heating will be terminated immediately.

KevinOConnor, 2019-01-07:
> When printing from Octoprint it will wait to send M112 until heating is complete.

That's odd.  That doesn't happen for me.  I just tested it by sending an `M109 S40` and then `M112` in the OctoPrint terminal tab - it stops immediately for me.

-Kevin

archer183, 2019-01-07:
Testing from the terminal  always works.   The fault occurs when printing
from octoprint and the print has m190.     It seems to not pass terminal
commands until m190 completes in that case.

KevinOConnor, 2019-01-07:
Huh - yeah - that failed for me too.  It's definitely octoprint though - if you ssh into the host and run `echo M112 > /tmp/printer` the printer stops immediately.

-Kevin

jakep82, 2019-01-07:
This seems like an issue that should be reported to the Octoprint developers.  It shouldn't lock out the terminal during heatup.  I don't have this issue because I always use virtual SD.  Another potential workaround would be to create a custom button that executes the echo command Kevin mentioned above.

KevinOConnor, 2019-04-04:
FYI, I opened an issue on the OctoPrint github to track this: https://github.com/foosel/OctoPrint/issues/3104

-Kevin

Nandox7, 2019-05-27:
I get this same issue but using RepetierServer and is not from the server code as when using Repetier Firmware the heating process is interrupted.

Either using the "Emergency stop" button from RepetierServer or manually sending M112 does not stop the heating process.

Nandox7, 2019-10-30:
Was Klipper code updated or the "fix" relies on a Octoprint change?
As I mentioned before this happens not only with Octoprint as I use RepetierServer and I have encounter this issue multiple times. Ending up having to cut power to the controller in order to stop it.