Klipper3d/klipper #1083: MCU timeouts after cold start

dothebart, 2019-01-06:
When I power up the raspii and the smoothieboard at once, the system runs unstable, gets `Lost communication with MCU 'mcu'` after some interaction with klipper, like G28. 

There are two ways to solve the situation 
 - (reliable) re-plug the USB-Connection to the smoothieboard
 - make klipper reset the MCU (if communication works out this seems to work too)

after that the communication works flawless.

dothebart, 2019-01-06:
[dmesg.txt](https://github.com/KevinOConnor/klipper/files/2730973/dmesg.txt)

KevinOConnor, 2019-01-16:
Strange.  I don't know why that would occur.  I'll add it to the list of things to take a look at, but I don't know when I'll be able to get to it.

-Kevin

dothebart, 2019-01-16:
is there anything I can do to get you more information about the state of the tty etc? Is this something that udev should handle?

dothebart, 2019-01-21:
is there some sort of command that could be used to get the protocol stack on the client into sync? I remember `^H` being used with modems to achieve that.

KevinOConnor, 2019-01-23:
Alas, I don't know enough about the root cause to guess what a solution might involve.  I don't even know enough to give any guidance on information to gather.  The only thing I can think of that I could do to help would be to try and reproduce it locally, but I don't know when I'll get time to do that.

-Kevin

dothebart, 2019-01-23:
ok, if ssh access to the pi can help, let me know. 
Funny is, that its knowing the git-id of the firmware of the MCU, right? so at least some communication takes place?

KevinOConnor, 2019-01-25:
The log indicates that the connection came fully up and was functional.  Then something externally closed the connection about 15 seconds later.  When the software did later reconnect (after manual intervention) the micro-controller was still running - so the mcu was not reset.

-Kevin

dothebart, 2019-01-28:
hm, ok. fresh boot: 
```
root@kossel:~# lsof |grep dev |grep 409
systemd-l 332                 root    7r      REG       0,15     4096       7654 /sys/devices/virtual/tty/tty0/active
python    409                   pi    0u      CHR        1,3      0t0       1028 /dev/null
python    409                   pi    1u      CHR        1,3      0t0       1028 /dev/null
python    409                   pi    2u      CHR        1,3      0t0       1028 /dev/null
python    409                   pi    3u      CHR        5,2      0t0       1109 /dev/ptmx
python    409                   pi    4u      CHR      136,1      0t0          4 /dev/pts/1
python    409                   pi    8u      CHR      166,0      0t0       7709 /dev/ttyACM0
python    409                   pi   12r      CHR        1,9      0t0       1032 /dev/urandom
python    409 468               pi    0u      CHR        1,3      0t0       1028 /dev/null
python    409 468               pi    1u      CHR        1,3      0t0       1028 /dev/null
python    409 468               pi    2u      CHR        1,3      0t0       1028 /dev/null
python    409 468               pi    3u      CHR        5,2      0t0       1109 /dev/ptmx
python    409 468               pi    4u      CHR      136,1      0t0          4 /dev/pts/1
python    409 468               pi    8u      CHR      166,0      0t0       7709 /dev/ttyACM0
python    409 468               pi   12r      CHR        1,9      0t0       1032 /dev/urandom
root@kossel:~# pstree -p |grep 409
           |-python(409)---{python}(468)
root@kossel:~# ps -eafL |grep 409
UID        PID  PPID   LWP  C NLWP STIME TTY          TIME CMD
pi         409     1   409  0    2 20:10 ?        00:00:01 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py /home/pi/printer.cfg -l /tmp/klippy.log
pi         409     1   468  0    2 20:10 ?        00:00:00 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py /home/pi/printer.cfg -l /tmp/klippy.log
root       965   877   965  0    1 20:22 pts/0    00:00:00 grep 409
```
Unplug, restart: 
```
[ ok ] Restarting klipper (via systemctl): klipper.service.
root@kossel:~# pstree -p |grep python 
           |-python(1016)-+-{python}(1035)
           |              |-{python}(1037)
           |              `-{python}(1038)
root@kossel:~# pstree -p |grep python 
           |-python(1016)-+-{python}(1035)
           |              |-{python}(1037)
           |              `-{python}(1038)
root@kossel:~# ps -eafL |grep 1016
UID        PID  PPID   LWP  C NLWP STIME TTY          TIME CMD
pi        1016     1  1016  2    4 20:34 ?        00:00:00 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py /home/pi/printer.cfg -l /tmp/klippy.log
pi        1016     1  1035  0    4 20:34 ?        00:00:00 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py /home/pi/printer.cfg -l /tmp/klippy.log
pi        1016     1  1037  0    4 20:34 ?        00:00:00 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py /home/pi/printer.cfg -l /tmp/klippy.log
pi        1016     1  1038  0    4 20:34 ?        00:00:00 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py /home/pi/printer.cfg -l /tmp/klippy.log
root      1044   877  1044  0    1 20:35 pts/0    00:00:00 grep 1016
root@kossel:~# lsof |grep dev |grep 1016 
python    1016                    pi    0u      CHR        1,3      0t0       1028 /dev/null
python    1016                    pi    1u      CHR        1,3      0t0       1028 /dev/null
python    1016                    pi    2u      CHR        1,3      0t0       1028 /dev/null
python    1016                    pi    3u      CHR        5,2      0t0       1109 /dev/ptmx
python    1016                    pi    4u      CHR      136,1      0t0          4 /dev/pts/1
python    1016                    pi    8u      CHR      166,1      0t0      13015 /dev/ttyACM1
python    1016                    pi   12r      CHR        1,9      0t0       1032 /dev/urandom
python    1016 1035               pi    0u      CHR        1,3      0t0       1028 /dev/null
python    1016 1035               pi    1u      CHR        1,3      0t0       1028 /dev/null
python    1016 1035               pi    2u      CHR        1,3      0t0       1028 /dev/null
python    1016 1035               pi    3u      CHR        5,2      0t0       1109 /dev/ptmx
python    1016 1035               pi    4u      CHR      136,1      0t0          4 /dev/pts/1
python    1016 1035               pi    8u      CHR      166,1      0t0      13015 /dev/ttyACM1
python    1016 1035               pi   12r      CHR        1,9      0t0       1032 /dev/urandom
python    1016 1037               pi    0u      CHR        1,3      0t0       1028 /dev/null
python    1016 1037               pi    1u      CHR        1,3      0t0       1028 /dev/null
python    1016 1037               pi    2u      CHR        1,3      0t0       1028 /dev/null
python    1016 1037               pi    3u      CHR        5,2      0t0       1109 /dev/ptmx
python    1016 1037               pi    4u      CHR      136,1      0t0          4 /dev/pts/1
python    1016 1037               pi    8u      CHR      166,1      0t0      13015 /dev/ttyACM1
python    1016 1037               pi   12r      CHR        1,9      0t0       1032 /dev/urandom
python    1016 1038               pi    0u      CHR        1,3      0t0       1028 /dev/null
python    1016 1038               pi    1u      CHR        1,3      0t0       1028 /dev/null
[... cut ...] 38               pi    2u      CHR        1,3      0t0       1028 /dev/null
python    1016 1038               pi    3u      CHR        5,2      0t0       1109 /dev/ptmx
python    1016 1038               pi    4u      CHR      136,1      0t0          4 /dev/pts/1
python    1016 1038               pi    8u      CHR      166,1      0t0      13015 /dev/ttyACM1
python    1016 1038               pi   12r      CHR        1,9      0t0       1032 /dev/urandom
```

So its starting more threads afterwards?

KevinOConnor, 2019-02-14:
I have not had a chance to look at this, but I did notice that @DanielJoyce is proposing something that may help - opening the serial port in exclusive mode.  It requires manually updating to the latest pyserial (as in pr #1251) and then manually applying the exclusive patch (commit fdbbda94).

-Kevin

DanielJoyce, 2019-02-14:
https://superuser.com/questions/569255/how-to-quickly-determine-which-process-if-any-listens-to-a-serial-port

Depending on which is installed you can use fuser or lsof to see what processes are using a file. But it may require you to get lucky and catch it when it happens.

Another option is possibly strace if available.

Check your printer configs. Do you have multiple printers? Are they accidentally using the same port?

Is power saving turned off for that usb port? This commonly puts serial ports to sleep.

dothebart, 2019-02-14:
no, the pyserial patch alone at least didn't fix it: [klippy.log](https://github.com/KevinOConnor/klipper/files/2866818/klippy.log)
(I only updated the host - the MCU is not necessary right?)

I only have one printer connected - the lsof doesn't find other processes opening the file - see the earlier post. 

I will try to find out about the power saving.

dothebart, 2019-02-14:
Running `udevadm info --export-db` before - just after the power up:
[before_udev_db.txt](https://github.com/KevinOConnor/klipper/files/2867020/before_udev_db.txt)
and after re-plugging:
[after_udev_db.txt](https://github.com/KevinOConnor/klipper/files/2867018/after_udev_db.txt)
...also no relevant changes when watched with meld - except that the device ids are counted up...

however, this gives a visible difference: 
`lsusb -D /dev/bus/usb/002/00[4|5]`

before:
```
  iManufacturer           1 Klipper
  iProduct                2 Klipper firmware
  iSerial                 3 12345
```
becomes:
```

  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
```
and this part is missing completely after the re-plug: 
```

Device Status:     0xffff
  Self Powered
  Remote Wakeup Enabled
  Test Mode
  Debug Mode
```
any hints if or what to change here to make them more similar?

dothebart, 2019-03-05:
Having had a discussion with another klipper user tonite who seems to have similar problems with his ramps-alike mks 1.3 board. His observation: the board powers up via the USB-Port, but doesn't get stable until the MCU power supply is there. 
In my case this could well be like that too; Though I power up the AC for the MCU and the PI  at the same time (the PI is behind a stepdown converter) maybe it reaches its 5v before the main MCU power is stable?

KevinOConnor, 2019-04-14:
I haven't had a chance to look at this.  The log indicates the usb device comes fully up, and is fully initialized by Klipper.  Then several seconds goes by and the connection is lost - it would be odd for power to be the root issue (unless the power was generally unstable).

Also, commit 7373e920 adds in exclusive locking on the serial port.  You should probably confirm the behaviour is still there with the latest code.

-Kevin

KevinOConnor, 2019-05-16:
Any further comments on this?  Did the latest code improve anything?

-Kevin

KevinOConnor, 2019-06-02:
I'm going to close this issue as it appears the conversation has concluded.  Let us know if the problem continues.

-Kevin

closed 2019-06-02 by KevinOConnor

KevinOConnor, 2019-07-12:
FYI, a report very similar to this one was tracked down to ModemManager running on the target machine.  Uninstalling ModemManager fixed the issue.

-Kevin

dothebart, 2019-07-15:
ah, thanks for letting us know, will try and report back!

dothebart, 2020-01-22:
`dpkg --list |grep -i modem` doesn't return anything - thus it seems as if no modem manager is installed

I tried sending USB-Reset via sysfs: https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line - no effect.

I finally found a "working" solution: 
daisychain  two of these: 
https://www.amazon.de/gp/product/B00BWF5U0M/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
The switches seem to disable the +5v USB-Lines - the D+/D- lines stay connected. So if you plug in an USB device powering itself they don't have any effect - the device will count as plugged in anyways, and the s-mini is plugged in via its own power suply. One also can see in the daisy chained setup that the hubs are only powering up a while after the pi is powered up ~3s.

The second non-self powered USB hub will only work with the +5v line being connected, thus cutting the power will disable it, and simulating a plugin/out event happening.  After that the connection is stable.

Starting with the second hub being powered from start doesn't help.