Klipper3d/klipper #1084: Klipper - Uninstall

PhilzzW, 2019-01-07:
Hi all

Please assist me with uninstalling Klipper. 

I have tried to sudo remove but it's not installed, it seems to be just 'running'. I stopped the service but with all of this, I can not seem to flash my boot loader with factory firmware.

I am stuck in a mess with a printer which is not working right now as I installed klipper only for the benefit of microcontrolling and speed purposes but I don't know what I'm doing and want it all back to factory.

Please help.

Thank you

yschroeder, 2019-01-07:
If available, just use the firmware update tool provided by your printer manufacturer and flash the recent firmware to it.

KevinOConnor, 2019-01-10:
A system uninstall script was recently added (commit 0ba1f510).  (You'd need to upgrade to the latest software to run this script - see the FAQ for details.)

As mentioned, once Klipper is stopped, you can flash another firmware.  You'd need to follow the flashing directions for that firmware.

-Kevin

closed 2019-01-16 by KevinOConnor

SixFiveLive, 2019-03-17:
Do i need to do anything (aside from flashing my firmware) after running the above script? Such as
 rm -rf ~/klippy-env ~/klipper ? I've run the script, but that line makes me think that is what takes the rest of the folders off the pi? I have no idea obviously. Or is just running the script enough, and not leaving anything to interfere in a future install? Thanks. I really need to get familiar with these environments.

AxMod3DPrint, 2019-03-18:
There is an uninstall script in the klipper/scripts folder run it as you
would the install.

Spartacus1066, 2019-03-23:
I know I need instructions like I don't know how to use a box of crayons, so if the answers above didn't get you, please do this. Type these two lines:
**sudo ~/klipper/scripts/klipper-uninstall.sh
rm -rf ~/klippy-env ~/klipper**

SixFiveLive, 2019-03-23:
Thanks. I was able to run it, I just didn't know what that last line it returned meant. You posting that command will surely help anyone that looks up the uninstall though.

jarjartee, 2019-06-03:
WHEN was the uninstall script added ??? Will this update be pulled in by apt-get update ??

theopensourcerer, 2019-06-03:
@jarjartee See kevin's earlier comment. It was added on the 10th Jan...

https://github.com/KevinOConnor/klipper/issues/1084#issuecomment-453175154