Klipper3d/klipper #1085: Filament Width Sensor Support

nopbit, 2019-01-07:
Hi All,
Is there any on going development for filament width sensor like in Marlin? If not i would like to develop one which steps i must follow? Sensor return DC voltage to analog pin (1.65v, 1.75v depends on filament's width) and according to that information i would like to change the flow rate. 
Thanks

nopbit, 2019-01-09:
Thanks for your answer Kevin I will read the documentation and check other codes. I have built one of those sensors on thingiverse https://www.thingiverse.com/thing:704897, it is based on this design https://www.thingiverse.com/thing:454584

Regards,

nopbit, 2019-01-11:
[filament_width_sensor sensor_name]
default_nominal_filament_dia = 1.75 # (mm)
measurement_delay_cm = 10 # (cm) The distance from the filament sensor to the melting chamber
pin = ^ar5

This is my configuration template. For example I have 10 cm distance from my sensor to melting chamber. I would like to read ADC on each 1 cm and store them in array.  I am planing to use first-in first-out (fifo) policy. When toolhead.epos reached to related position i will change the extruder factor with  M221 according to readings. 

I have a question about events. Is there any event that is triggered on each extruder stepper movement/step motor movement or i have to use reactor.register_timer? 

Regards

KevinOConnor, 2019-01-13:
There isn't any generic event generated on each extrusion move.

If you are measuring simple ADC values, it should be possible to instruct the micro-controller code to measure and transmit an ADC value at a fixed time interval (eg, every 500ms).

I suspect generating an M221 would not work well.  I suspect a more flexible solution would involve updating the extruder kinematics with any upcoming filament adjustments.  This would allow the extruder code to apply the adjustments without any delays introduced by the lookahead queue.

Cheers,
-Kevin

nopbit, 2019-01-14:
`self.gcode = self.printer.lookup_object('gcode')`
`self.gcode.extrude_factor = new_extrude_factor`
will this code effect directly to extruder kinematics without any delay?

Regards,
Mustafa

KevinOConnor, 2019-01-14:
It would apply the change to all future g-code commands that are read.  It would not apply the change to any g-code extrude commands that have already been read and queued in the look-ahead queue.  There is some information on the look-ahead queue at: https://github.com/KevinOConnor/klipper/blob/master/docs/Kinematics.md

-Kevin

KevinOConnor, 2019-02-08:
I'm going to close this issue as I think PR #1132 is a better place to track.

-Kevin

closed 2019-02-08 by KevinOConnor