Klipper3d/klipper #1087: Cetus 3D - sensorless homing

etet100, 2019-01-07:
Hello. Sorry for my poor english.

I'm going to try to replace Cetus 3D board with klipper / stm32f103 based board. The problem is that Cetus MK2 has no mechanical endstops. It uses simple circuit to detect electrical spikes from motors. From the software side it's just waiting until endstop pin state is low for 1 to 1.5 ms (or something like this). Every high state (even shortest one) is resetting timeout. 

Do you think it's possible to modify endstops to work described? Please let me know if something is not clear to you.

etet100, 2019-01-15:
Ok. Thanks. I think it's not possible without using interrupts on endstop pins.

KevinOConnor, 2019-01-27:
Okay, I'm going to close this issue then as it looks like the original question was answered.

-Kevin

closed 2019-01-27 by KevinOConnor