Klipper3d/klipper #1088: One axis shuts down at specific line of print

Iakabos, 2019-01-08:
I'm having a problem where a specific line of gcode in one print (sliced with cura) is causing my B motor driver (coreXY system) to shutdown. I am running TMC2208s but it doesn't appear to be the same as issue #196 since I'm running the drivers in 'dumb' mode (no UART) and it's not random. The last 20 or so files I've printed have been fine, and it happens reliably at the same line in this file every time.

Here's the log: [klippy1.log](https://github.com/KevinOConnor/klipper/files/2735484/klippy1.log)
And here's a trimmed version of the print gcode: [Driver_failure_test.txt](https://github.com/KevinOConnor/klipper/files/2735486/Driver_failure_test.txt)

Specifically, if I comment out the line: G1 X240.076 Y107.988 E1709.90273 it carries on as normal, but with that line enabled, it shuts down the driver every time. I've confirmed it's not overheating, and I see nothing glaringly wrong with that line. I'm assuming this is an issue with how klipper handles that line?

Sorry if I forgot any details. Please let me know if you need any further information.

theopensourcerer, 2019-01-08:
This sounds exactly like #196 in that it is when 2208s are used in standalone mode. I had a similar issue but once I'd configured them for UART control the problem went away.

Iakabos, 2019-01-09:
Okay. I had understood that 196 was specifically a UART mode issue but if I had it backwards and that actually solves the problem I'll have to wire that up.

KevinOConnor, 2019-01-21:
I'm going to close this issue - lets track in #196.

-Kevin

closed 2019-01-21 by KevinOConnor