Klipper3d/klipper #1090: TB6600 support (or stepper timing differences)

Coolaswa, 2019-01-08:
I have some TB6600 drivers, which clunk sometimes, shifting an axis thus ruining a print. I need because I have a relatively large printer thus a large torque. These TB6600 are relatively slow chips, and Marlin compensates for this [see here](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L819-L855). 
I would like to request similar options in klipper. Or if someone could point me to how to do it would also be appreciated.

Regards

KevinOConnor, 2019-01-09:
This has come up a few times before.  I updated the configuration process so that one can configure the step pulse delay during "make menuconfig" (commit 2ea7c97b).  You'll need to pull down the latest code (see the FAQ for details), rerun "make menuconfig", enable the "low level config options menu", and then customize the step pulse duration (looks like you need 3us).

-Kevin

Coolaswa, 2019-01-10:
Is the decreased steps the cause of the 'Timer too close' errors I am getting? I decreased velocity by 3x, and I keep getting these errors.

KevinOConnor, 2019-01-10:
You'll need to attach the klipper log file containing the event in order for us to assist.  See:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

Coolaswa, 2019-01-11:
Off course, sorry. This is the result after an xy home, and some 10 cm move in x direction. Speeds have already been tuned down, but even if I go down to 10 mm/s the error remains. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/2749194/klippy.log)

KevinOConnor, 2019-01-11:
Alas, there was a bug in the stepper code when using a non-zero step delay on AVR chips.  It should be fixed now (commit 8a6125c8).  You'll need to pull the code again, recompile the micro-controller code and reflash to get the fix.

I suspect that 100mm/s is going to be too fast for your setup - setting a custom step delay on the AVR disables the AVR stepper optimizations.

Separately, if using a multi-mcu setup, I highly recommend you follow the steps at: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

-Kevin

KevinOConnor, 2019-01-11:
FWIW, if you just need to go to 3us you could try setting the step delay back to zero and comment out the line `if (s->flags & SF_HAVE_ADD)` in src/stepper.c.  Then recompile and reflash.

-Kevin