Klipper3d/klipper #1093: [Suggestion] Proper source folder naming

FHeilmann, 2019-01-08:
Since klipper contains multiple src subfolders that cater to a specific architecture as opposed to a specific mcu I suggest renaming them in a uniform way.

- avr devices are covered by the folder `avr`
- atsam devices like the sam3 and sam4 family are currently in a folder `sam3`, I suggest renaming this folder to `atsam` as that is how most libraries call this family of mcus
- the samd21 support is currently in a folder `samd21`. Since samd21 encompasses multiple mcus, and a very similar architecture exists in the samd51 family of chips which will probably become very relevant for klipper once the Adafruit Grand Central is released, I suggest renaming this folder to `atsamd`

Cheers,
Florian

KevinOConnor, 2019-01-09:
Interesting.  I didn't realize the SAMD21 peripherals are also in higher end chips (eg, the SAMD51).

If you'd like to submit the above changes, I'm fine with it.

-Kevin

FHeilmann, 2019-01-10:
See #1100

closed 2019-01-10 by FHeilmann