Klipper3d/klipper #1095: Using IR Z Probe 1.4 with Klipper

lmcbmai, 2019-01-08:
Hi, 
Has anyone installed and using with Klipper 'IR Z Probe 1.4' bed leveling sensor from https://aus3d.com.au/ir-z-probe ? If yes, could you please share your experience and configuration settings? 

Thanks
Best regards,
Boris

kingfisher1234, 2019-01-09:
I'm using a "duet" one (https://miscsolutions.wordpress.com/mini-height-sensor-board/)

in the Z stepper section change the following

endstop_pin: probe:z_virtual_endstop
#position_endstop: 0
position_min: -3

In the probe section: 

[probe]
pin: ^P4.28
x_offset: 0
y_offset: 0.0
z_offset: 0.0
speed: 5.0
#activate_gcode:
#deactivate_gcode:

NB:
My pins are for a cloned smoothie board so you will need to change these
leave the activate/deactivate sections commented out as there is no servo to call
Remember to change endstop_pin to "probe:z_virtual_endstop" (This caught me out)
If you get out of range errors lower position_min
I added a Z offset in my startup macro rather than in the probe settings (e.g "G92 Z1.55") But you can add this above instead

lmcbmai, 2019-01-10:
@kingfisher1234 Thanks for your reply. I'm expecting the sensor to arrive in about a week. Will try it and post the update here.

KevinOConnor, 2019-01-21:
I'm going to close this issue as it doesn't look like anyone here has that exact sensor (though others are using other IR sensors).

-Kevin

closed 2019-01-21 by KevinOConnor