Klipper3d/klipper #1096: SKR 1.1 with lpc1768 Thermistor ADC issue?

derdigge, 2019-01-09:
Dear Kevin.

As i am running a config for a CR10 with theese 32 bit board, i wanted to pull here see #1010 , i am facing a strange problem.
My Thermistor of the Hotend is bumping arround. It is a 24V board with Meanwell LRS-200-24 as powersupply. I did allready the usual stuff like checking the wires connectors and such. Its a .25mm² wire for the thermistor.  

For testing i applied a short Wildwire with a testhotend close to the Board to reduce the influences here. (with 0.25mm² and 0.5mm² wires) Also i tried to check the real temperature as good as i can with a laser thermometer. I was close to octoprints shown temp. Allways about 15°C lower with laserthermometer.

What can i do? Why is this bumping so hard? Is it a bad hardware taste?

Thanks 
derdigge
[klippy.log](https://github.com/KevinOConnor/klipper/files/2739899/klippy.log)

derdigge, 2019-01-09:
the thermistor has 3.3v

derdigge, 2019-01-09:
Update. I installed smoothie on it and the thermistor is working good with reasonable values. So we are doing something wrong for that board on klipper.

KevinOConnor, 2019-01-09:
This looks like a dup of issue #842.  It is believed that PR #1086 fixes the issue.

-Kevin

derdigge, 2019-01-09:
Thanks i try that.

closed 2019-01-09 by derdigge