Klipper3d/klipper #1101: MCU not shutting down heaters after connection loss

Kevin-No, 2019-01-10:
Hello!

In a longer print yesterday Klipper lost connection to the MCU (MKS Gen L 1.0). Couldn't find the reason yet, good PSU, no power warnings from OctoPi, common ground, good quality short USB cable [...]. However that's not the problem which is actually that the printer was just sitting there with heaters still on when I came to check it 1.5 hours after connection loss.b Reconnecting to the MCU worked fine after repowering the MCU (was killing power in an overreaction).

If I didn't misread the docs the MCU is supposed to turn of heaters after an extended time of disconnection (5s?). As the temperature seemed to stay at printing temperature this might only be a mild safety concern but what bothers me is the question if sanity checks are performed by the MCU while Klippy is disconnected or might it end up in a bad state with fully powered heaters? Or are these Checks even trustworthy if it already failed to turn off heaters on disconnect?

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/2744650/klippy.log.zip)

KevinOConnor, 2019-01-11:
That's strange.  Nothing in the log looks unusual (other than the disconnect).  I recommend you re-run the M112 steps in the config checks document: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

That M112 test is intended to verify that the heaters go into the correct state on a firmware "shutdown" state.  It's hard to verify that the micro-controller did enter the shutdown state (or if it was operational at all) during the event.  It's easy to test the general shutdown capability though - just set a temperature and then unplug the usb cable to the micro-controller - you should see it turn off the heaters in about 5 seconds.

Finally, just for your reference, the board disconnected a few seconds after the bed heater turned on.  Using the watermark heater in the bed heater section causes the bed mosfet to turn on for extended periods.  On low-end "ramps" and similar boards, this can cause heat buildup.  Using the PID and using a setting like max_power=0.5 can mitigate that.

-Kevin

Kevin-No, 2019-01-11:
Hi Kevin,

thanks for your reply. Unfortunately I have not been able to reproduce this issue. M112 worked like discribed in https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md and unplugging the cable caused the heaters to shut down within a few seconds of the event according to the indicating LEDs. If I can do anything else to check for possible failures or if there is a way to produce a more helpful logs I'd be happy to try it out.

And thanks for your comment about the watermark heater. I didn't bother to change it to Pid and tune it yet as I'm powering the bed with  beefy up mosfet and heatsink in an actively cooled enclosure. But I'm going to keep an eye on that and get Pid running sooner or later anyways.

Kevin-No, 2019-01-11:
Just realized that the timestamps in Octoprint are off by an hour so yeah, I was cutting power about half an hour after the disconnect. Which leads me to think that the disconnect might actually have something to do with me blindly hitting things on my phone while the alarm rang. I did check out Printoid just before going to sleep so the app might have still been open when I desperately tried to stop the alarm still 90% asleep.

Both LEDs where on, Bed defenitly felt like printing temperature at 80C, didn't check the hotend for obvious reasons.

Kevin-No, 2019-01-18:
Update: still have not been able to reproduce this, not even with wild and random tapping in printoid. Didn't get it to disconnect. Also didn't have any random disconnect. So if you want to pass this as a really odd very corner case feel free to close this issue. I'll leave it opened to leave the decision up to you.

Thanks for your time and awesome work on this project,
Kevin

KevinOConnor, 2019-01-27:
Thanks.  Lets close this then as I don't see any way to further investigate it.

-Kevin

closed 2019-01-27 by KevinOConnor