Klipper3d/klipper #1103: SKR TMC Uart is failing when extruder is present in config.

derdigge, 2019-01-11:
Hello Kevin.

During creating the config for my voron 2.1 i stombled into a interesting issue.
If i remove the extruder section alltogether from config the printer works. I can home and travel arround.
For checking my mechanics it was on travel for an hour with different microsteps and currents i set through uart. All my 7 drivers are tmc2208 with uart on two SKR 1.1(smoothi like) boards.

But when i add a extruder config it dies with:

´´´ERROR:root:Unhandled exception during connect
Traceback (most recent call last):
  File "/root/klipper/klippy/klippy.py", line 140, in _connect
    cb('connect')
  File "/root/klipper/klippy/extras/tmc2208.py", line 197, in printer_state
    self.set_register(reg_name, val)
  File "/root/klipper/klippy/extras/tmc2208.py", line 217, in set_register
    self.ifcnt = self.get_register("IFCNT")
  File "/root/klipper/klippy/extras/tmc2208.py", line 203, in get_register
    [self.oid, msg, 10], 'tmcuart_response', self.oid)
  File "/root/klipper/klippy/serialhdl.py", line 189, in send_with_response
    return src.get_response()
  File "/root/klipper/klippy/serialhdl.py", line 223, in get_response
    raise error("Timeout on wait for '%s' response" % (self.name,))
error: Timeout on wait for 'tmcuart_response' response
´´´´

See klippy log. I deletet klippy log and made only one startup to keep it clean. This happens on lastest Master andd also on work-median-filter-20190104. So this is missleading and took me 2 hours to realize that it does shutdown because of extruder temp out of range. 

Maybe you know the place to handle this case for the next lonly soul? Something like extruder warning overcomes any othe failure during startup?

Thanks and BR
derdigge
[klippy.log](https://github.com/KevinOConnor/klipper/files/2749062/klippy.log)

KevinOConnor, 2019-01-11:
I pushed a change that should hopefully make this a little more clear (commit f0b7d6d6).

-Kevin

KevinOConnor, 2019-01-25:
Hopefully this is fixed now.

-Kevin

closed 2019-01-25 by KevinOConnor

derdigge, 2019-01-27:
true it works now.