Klipper3d/klipper #1104: Usb spontaneously changing

BearInThere, 2019-01-11:
OK first things first 
1. I am not using Klipper under the normal circumstance
2. To me it appears this is a RPI OS issue.

I'm currently writing a Windows App that can  communicate with 3D firmware ie. Klipper, Marlin etc.
To communicate with Klipper what I did was install 'Ser2Net' targeting 'tmp/printer' on my RPi and ensured that OctoPrint is not connected.

This works perfectly, but I have noticed if left in an idle state for an hour or two 
 ../../ttyUSB0 
will change to
 ../../ttyUSB1
spontaneously

Klipper works correctly so this is obviously an OS  issue  but I am hoping someone here has some insight. 

Yes I did read the FAQ

Cheers
Dan

BearInThere, 2019-01-11:
I apologise for this as I do not believe it is an issue with Klipper.

0rinsb3lt, 2019-01-12:
This can be fixed via udev rules
I personally put mine in `/etc/udev/rules.d/99-custom.rules`
essentially, it'll make a symlink to the ttyUSB{x} device
`ACTION=="add",SUBSYSTEM=="tty",ATTRS{idVendor}=="{VendorID}",ATTRS{idProduct}=="{ProductID}",SYMLINK+="{PrinterName}",GROUP="dialout"`

change ProductID, VendorID, and PrinterName
PrinterName can be anything you want.

The code above is all on 1 line.

easiest way to find vendor and product id
1) remove usb cable to the 3d printer
2) execute `lsusb`
3) plug in usb cable to 3d printer
4) execute `lsusb` (again)

The device that is different between the 2 runs will be your usb adapter that connects
to the 3d Printer. Use that ID XXXX:XXXX for VendorID and ProductID. The format 
lsusb outputs is ID VendorID:ProductID.

using the PrinterName you choose above
in your `printer.cfg` you will have to change from `/dev/USBx` to `/dev/{PrinterName}`

Now anytime the 3d Printer reconnects to the rPI it'll use the same device symlink
and you'll only have to issue a `firmware-restart`

This method has 1 draw back though. 
If you have more than 1 device that uses the same vendor:product id on the tty subsystem it could mix them up.

Hope the explanation is thorough enough to help.

BearInThere, 2019-01-12:
Thanks that should help
At the moment I'm using 1 x 3Drag clone board but I do intends to add 2xSTM32 Bluepills.
At least now I'm expecting it so I'll deal with that as it arises.

Cheers

jakep82, 2019-01-12:
Did you see this section in the FAQ?

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

0rinsb3lt, 2019-01-12:
Personally I didn't look in the FAQ either. That solution is a lot better than udev rules.
Thank you for pointing it out!

BearInThere, 2019-01-12:
Did you see the bit were I wrote 'Yes I did read the FAQ'
So yes I did read that but it only talks about locating the current USB device.

>ls -l /dev/serial/by-id/
>lrwxrwxrwx 1 root root 13 Jun  1 21:12 usb-1a86_USB2.0-Serial-if00-port0 -> ../../ttyUSB0

I'm talking about the port spontaneously changing which results in a message from Klipper:

// Klipper state: Shutdown

then when I type in 'Status' I get

// Lost communication with MCU 'mcu'
// Once the underlying issue is corrected, use the
// "FIRMWARE_RESTART" command to reset the firmware, reload the
// config, and restart the host software.
// Printer is shutdown
!! Lost communication with MCU 'mcu'

then

>ls -l /dev/serial/by-id/
>lrwxrwxrwx 1 root root 13 Jun  1 21:12 usb-1a86_USB2.0-Serial-if00-port0 -> ../../ttyUSB1

Notice it has changed to '../../ttyUSB1' the second time.
I haven't had a chance to try Orinsb3lt solution yet but hopefully since it would be now using  '/dev/{PrinterName}' it shouldn't change.

BearInThere, 2019-01-12:
Also, if I disconnect and reconnect the USB it makes no difference. Bit if I
>disconnect usb
>sudo service klipper stop
>reconnect usb
>sudo service klipper start

then it seems to correct the issue and I'm back on '../../ttyUSB0'

jakep82, 2019-01-12:
You should read it a little closer. If you follow the instructions, you won't have the problem you're describing.  The by-id serial name is not volatile.

BearInThere, 2019-01-12:
jakep82 thanks I've read it a few times but your persistence made me look a little bit closer.
It turns out that I'm using an edited version of
 'printer-velleman-k8200-2013.cfg ' 
which contains
>[mcu]
>serial: /dev/ttyUSB0

I've changed that to
>[mcu]
>serial: /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0-port0

I'll let it sit for a few hours but based on your comment 'The symlink (aka /dev/ttyUSBX) is volatile.' all should be good.

I apologise for my ignorance of Linux and thank you for your help.

jakep82, 2019-01-12:
You shouldn't have any more issues.  Both by-id and by-path should be stable.

BearInThere, 2019-01-13:
Unfortunately this has improved the situation but not fixed the problem. So I just want to clarify a couple of things.

First Since I am now using 
>[mcu]
>serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A702TNGK-if00-port0
in my config it makes no difference whether the port gets named '../../ttyUSB0' or'../../ttyUSB1' and this appears to be the case since Klipper will connect either way.

Second, my understanding from what I have read is that once the MCU is plugged into the RPi what ever name it is given should not change until it is disconnected and reconnected, and then it's random.

If my second point is correct then does that suggest that I maybe looking for a physical issue not a config / software one? Since on the occasions when Klipper does shut down because it has lost communication with the MCU the name of the USB port has also changed.

cmock, 2019-01-15:
I've had this problem with unstable power to the MCU board; at some point, the USB chip there resets, which leads to this renaming business... so I'd suggest checking your power setup.

BearInThere, 2019-01-15:
Yeah thanks, I did read that in FAQ and am waiting on some new USB micro connectors, my temp fix should hold 'till then.