Klipper3d/klipper #1105: Servos/BLTouch Not working on Replicape

chappo, 2019-01-12:
Hi Guys,

Having issues getting the servo's working on the latest build of Debian Jesse. Using universalh overlay as I am using the emmc.

All x,y,z steppers, heating is working etc it's just the servo's.

Servo support is enabled in the config file. Also tried configuring a "servo" in the config with no success. 

Output of config pin below - 

config-pin -q P9_11 = P9_11 Mode: default Direction: in Value: 0
config-pin -q P9_14 = P9_14 Mode: default Direction: in Value: 0
config-pin -q P9_28 =P9_28 pinmux file not found!  / Cannot read pinmux file: /sys/devices/platform/ocp/ocp*P9_28_pinmux/state
config-pin -q P9_16  = P9_16 Mode: default Direction: out Value: 0
config-pin -q P9_25 = P9_25 Mode: default Direction: in Value: 0

dmesg | grep 'pin' =
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 129408
[    0.455988] pinctrl core: initialized pinctrl subsystem
[    0.664352] omap_i2c 44e0b000.i2c: could not find pctldev for node /ocp/l4_wkup@44c00000/scm@210000/pinmux@800/pinmux_i2c0_pins, deferring probe
[    1.178723] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[   32.457672] pru-rproc 4a334000.pru0: stopping the manually booted PRU core
[   32.458413] pru-rproc 4a338000.pru1: stopping the manually booted PRU core

chappo, 2019-01-12:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2751772/klippy.log)
[uEnv.txt](https://github.com/KevinOConnor/klipper/files/2751773/uEnv.txt)
[version.txt](https://github.com/KevinOConnor/klipper/files/2751774/version.txt)

jakep82, 2019-01-12:
I'll be curious to see if you get this working.  I spent quite a few hours fighting this exact issue, and I eventually gave up.  

Are you sure your steppers are fully functional?  For me the issue was the shift registers which I believe are controlled by SPI.  When I used an overlay that gave me functional heaters and steppers, the shift register didn't work so I was unable to change my microstep settings.  The shift register is also necessary to make the servos work.

When I used an overlay that gave me a functional shift register, my bltouch worked as expected.  Unfortunately this didn't give me proper pin configurations for my steppers, so they stopped working entirely.  I was unable to find any configuration in /boot/uEnv.txt that gave me a fully functional printer.

chappo, 2019-01-12:
> Are you sure your steppers are fully functional? 

No I am not 100% confident. Steppers home etc just fine. I haven't tried changing the micro-stepping.

> I spent quite a few hours fighting

I am up to about 15 hours of trying various things to get it to work, multiple versions playing around with the cape manager uboot etc.

BlackStump, 2019-01-12:
I have and use a Replicape on a Delta Printer but never needed to enable/test the servo's but am curious as to why they dont work for you.
what commands are you issuing to read the pins ?
[uEnv.txt](https://github.com/KevinOConnor/klipper/files/2751812/uEnv.txt)

KevinOConnor, 2019-01-21:
FWIW, this has been a continual problem with the Beaglebone - the mechanism to configure the state of the gpio pins has not been stable.  At this time it is my understanding that the Klipper code itself is correct, but the mechanism to configure the OS to allow access to the hardware is not known.  If someone wants to track this down and help us document then that would be great.  I'll add it to my list of things to look at, but I don't have a timeframe to accomplish that.

-Kevin

chappo, 2019-01-21:
Thanks Kevin. 

What is klipper expecting in terms of BBB configuration? Cape Manager? U-boot? with overlays mapped manually or automatically? Cape Universal enabled?

Looking through the code it seems SPI is in use for the Servo's? Which SPI number is it expecting? Is it a matter of changing bit-register and rebuilding klipper?

Do we need to create a udev rule like they have done with kamikaze?
https://github.com/eliasbakken/Kamikaze2/blob/v2.0.6/scripts/spidev.rules 

Everything else appears to be working OK. It is just the servo's which are not working.

jakep82, 2019-01-21:
Without SPI working, I don't think it's possible to properly configure stepper drivers either.  I know when I was trying to make it work, changes to microstep settings were not applied.

chappo, 2019-01-23:
May be a simple fix after all - the replicape config file seems to reference incorrect pin numbers for the servo's.

From replicape.cfg
"or for servo_0 (via P9_14)."
"or for servo_1 (via P9_16)"

From - https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-BONE-REPLICAP-0B3A.dts
"P9.16", /* SERVO_0  */
"P9.18", /* SERVO_1 */

jakep82, 2019-01-24:
I noticed this too when I was trying to get this working, but I think this is actually a mistake in the dts file.  Pins P9_14 and P9_16 are referenced for the servos in a couple places further down the dts file.  I edited the dts file and compiled a new dtbo overlay with fixed pin definitions, but it made no difference for me.

chappo, 2019-01-27:
Thanks Jake. One thing that I have noticed is that when you don't have the bltouch in the config the red and blue leds are on. When the bltouch is enabled only getting a red led - it also doesn't matter if I have the command pin set to P9-14 or P9-16 the led behavior is the same.

jannau, 2019-03-23:
The problem is that P9_14 and P9_16 are pin-muxed to the SoC's hardware PWM unit. They are not available for software PWM in the PRU.
I've added support for the sysfs pwm interface to the linux host mcu. It seems to work, only tested pin_down/pin_up debug commands so far. I'll prepare a merge request after cleaning my changes up.

jannau, 2019-03-24:
See #1443.
Controlling the BLTouch works but the endstop is never triggered.

closed 2019-04-06 by KevinOConnor

KevinOConnor, 2019-04-06:
FYI, this should now be fixed in the latest code (commit 478a916).

-Kevin

jannau, 2019-04-06:
It works but needs config changes. servo0_enable/servo1_enable are not longer used and need to be removed from the config. The servo pin or bltouch control pin needs to be be either replicape:servo0 or replicape:servo1. See the changes to [generic-replicape.cfg](https://github.com/KevinOConnor/klipper/config/generic-replicape.cfg) in https://github.com/KevinOConnor/klipper/commit/478a916f516fa05aea3b4e94cfe04fc10786c191

chappo, 2019-04-08:
Thanks Jannau / Kevin. 

In the process of testing at the moment. Can you please confirm the wiring. Do we need to use the X2 connector for the servo / ground and then Y2 as the end stop? I can see that both P9_11, P9_14 are reserved for servo0.

chappo, 2019-04-08:
Hi Jannau,

Can you please confirm a) debian version b) kernel version c) uEnv config.

The beaglebone instructions aren't very clear in what works.

jannau, 2019-04-08:
Debian Stretch using the TI kernel 4.4 (currently 4.4.155-ti-r153). Only modification in uEnv.txt is enabling the matching TI pru  overlay like:
```
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
```

commit 478a916f (2019-04-06): replicape: add support for Linux hardware PWM (#1443)

The servo pins (P9_14/P9_16) are muxed to the SOCs hardware PWM unit
driven by a 13MHz GP timer. They have to be driven by the linux host
mcu. This commits adds hardware PWM support using the linux sysfs
user space interface.

The servo pins can be specified as "replicape:servo0" and
"replicape:servo1". Removes the "servo0_enable", "servo1_enable"
configuration parameters.
Fixes #1105.