Klipper3d/klipper #1112: Failure at 91% after 3 hours of printing

lmcbmai, 2019-01-14:
Hi,
I am still very new to Klipper. So far I only printed few small test items (1 hour-long prints or less). The results were perfect even at speeds up to 160mm/sec and today I decided to try a longer print. 
After 3 hours at 91% completion the print failed.  The error message is exactly the same as in [Issue 392](https://github.com/KevinOConnor/klipper/issues/392). 
I am attaching zip file that contains klippy.log, printer.cfg, octoprint.log and the gcode file I was printing.
Print speed was set to 120 and wall speed to 60.  This particular item above layer 4 contains only walls, so the failure happened at speed 60. 

I should be running the latest Klipper code, as I updated it yesterday.

Any advice on how to avoid similar failures in future will be very much appreciated. 

[Print_Failure.zip](https://github.com/KevinOConnor/klipper/files/2754204/Print_Failure.zip)

Thanks
Best regards,
Boris

lmcbmai, 2019-01-15:
Here is the chart. Can anyone please tell if it tells what caused the failure?
[image]

Thanks

KevinOConnor, 2019-01-16:
A "move queue empty" error indicates that the host and micro-controller got out of sync - either a buffer management error or the host was unable to make an accurate prediction of the micro-controller clock.  It looks like you've made changes to the Klipper code.  For this type of error, it would be necessary to reproduce it on the pristine code for me to be able to provided more detailed assistance.

-Kevin

KevinOConnor, 2019-01-16:
Separately, if you're looking to debug this yourself, take a look at the regular graphstats output (`graphstats.py klippy.log load.png`) - the graph indicates the host is highly loaded during the prints.  (The "host buffer" reports under normal circumstances should be very low.)

-Kevin

lmcbmai, 2019-01-25:
@KevinOConnor 
Got "move queue empty" error again today. At this time I am on "clean" Klipper code. Print failed at 6%. Both - extruder and bed heaters switched off (which is good), however OctoPrint GUI hang the browser at such a level that I could not even close Chrome tab showing OctoPrint.  After restarting the browser OctoPrint page loaded and showed that it keeps printing (sending commands to the port); print completion was over 50%. Shortly after, the browser became unresponsive again. While this was happening, 'top' printout showed that OctoPrint CPU utilization is at 115% (!!!). 
[image]
I was printing with PLA at 0.32mm layer height. Print speed was set to 75mm/s, travel speed was 120mm/s. During initial testing, I printed simple objects with speeds up-to 160mm/sec and had no issues, i.e. this is not a problem with kinematic not being able to perform. I checked CLOCK_MONOTONIC_RAW that helped to solve similar issue in the past. It is now present in pyhelper.c file. 

Today's failure happened when, based on the following chart, neither host, nor MCU appear to be overloaded.
[image]

Here is MCU frequency:
[image]

And here is the archive with klippy.log and gcode file that failed to print:
[Print_Failure.zip](https://github.com/KevinOConnor/klipper/files/2797564/Print_Failure.zip)

Looking forward to resolve this issue, as I really don't like the idea of going back to slow printing Marlin.

Thanks
Best regards,
Boris

lmcbmai, 2019-01-26:
Couple of additional observations and thoughts. During the last failure I was printing a box shaped object. It has 4 bottom layers followed by 73 wall layers. The failure happened right after completing the first 4 layers, i.e. during transition from bottom to walls. Load utilization chart shows 4 MCU utilization spikes that most likely happen at layer change.  Every new spike was higher than the previous (I wonder why ?) and the last spike could have exceeded 70%. Could this be  the reason of the failure? I have several other chart of successful prints - none of them has MCU utilization above 50% during the print. They do have spikes up to the values close to 250, but that did not cause any problem.
My [stepper_z] step_distance is set to .000234375. The value in the original Marlin-based firmware was 4266.66 steps/mm (all motors run in 1/32 microstepping mode).  Can this cause an issue? What would be cons and pros switching off microstepping on Z motor? Would that reduce load and the amount of calculations done by the host? 
Thanks

lmcbmai, 2019-01-27:
Hi Kevin, 
Thank for you reply. I was not ready to go all the way and reinstall entire OctoPrint (my installation is just about 2 months old), but I switched off all the plugings, Octoprint serial log and even camera. Under such bare-bone setup I was able to successfully print several parts with total print time about 12 hours. I was really hopeful that the problem is gone, but today I decided to give it a try to the box shaped part that failed last time. It made it through half of the layers and failed again. I am attaching the log and the chart, in case you want to look at them. 
[image]
[klippy.log_27-jan-2019.zip](https://github.com/KevinOConnor/klipper/files/2800572/klippy.log_27-jan-2019.zip)
As I've mentioned earlier, I switched to Klipper about 4 weeks ago hoping to increase printing speed and also to try new firmware.  Based on my experience so far, I am getting impression that there are 2 main problems with having host and MCU running separately :
1) lack of 2 way communication between MCU and Host that causes timing issues, which turn into print failures
2) dependency of host software on other processes running on Raspberry and no easy way to determine how much, or how little room is available on Raspberry for adding additional features/plugins to Octoprint. 

From one side I'd love to stay on Klipper, but I have several projects that started to be affected by the issues I am having with my printer. I will keep trying for few more days and if no improvement, will have to switch back to Marlin,

Thanks
Boris

jakep82, 2019-01-28:
I think your Z axis is overloading your MCU.  With your current Z step settings you need about 4260 steps to move 1mm.  You have a max Z velocity of 25mm/s.  At that speed you would need more than 106K steps per second which is more than your mega2560 can handle.  I would suggest reducing your max Z speed to no more than 10mm/s, or consider changing the microstep settings on your Z axis to increase your step distance.

lmcbmai, 2019-01-28:
@jakep82  Actually, you are probably right. I just pulled out the original Marlin's _configuration.h_ file that came with the printer and it has 10mm/s for Z-axis. Looks like I missed this setting and kept the default from _example-corexy.cfg_ file. Decided to lowered it to 5mm/sec, to be on the safe side.  Hopefully it helps.
Thanks!
-b

KevinOConnor, 2019-01-28:
It's unclear what the root cause of this latest error is.  It is a different error (a "rescheduled timer in the past") from previous reports.  This error generally indicates a request to step at a rate higher than the micro-controller can handle.  The maximum Z step rate does look high, but it does not seem like it is the root cause of this issue as the log does not seem to indicate that there was Z movement occurring at the time of the fault.

I do not believe Klipper has a systemic issue with host scheduling.  I've not received any other similar reports; neither of the errors reported ("move queue empty" / "rescheduled timer in the past") are indicative of host scheduling problems; the quoted snippet from top shows the rpi as a whole at 66% idle.

There does appear that something is not correct on your particular setup.  It's unclear what it is.  I can think of some exotic reasons (such as a failing crystal on your micro-controller board), but these seem unlikely explanations.  You've mentioned that the same print has failed twice - if you can attach the exact gcode from that print I can try to simulate it here to see if I can reproduce the problem.

-Kevin

KevinOConnor, 2019-01-28:
> Writing to disk or even flash memory is SLOW compared to everything else the CPU does.

All log messages in Klipper are written to an internal memory queue and a dedicated background thread writes them to disk.  There is no indication that logging itself is contributing to this failure.

Separately, the latest error log is indicative of a failure originating at the micro-controller; it is unlikely that a "rescheduled timer in the past" error would result from a problem on the host.

-Kevin

lmcbmai, 2019-02-03:
Quick update: No single failure since my last post. Looks like adjusting z step setting helped. Right now all OctoPrint plugins are still disabled. Camera is enabled. My next step will be enabling plugins one by one to find out if any of them is causing troubles. I will update this thread with my findings.

lmcbmai, 2019-02-11:
Just successfully completed 10 hours-long print with all add-plugins enabled. Looks like the problem was caused by too high speed setting along Z-axis and after lowering it the issue is gone. Many thanks to everyone who replied to this topic. I am closing this ticket.

closed 2019-02-11 by lmcbmai