Klipper3d/klipper #1113: Automatic mesh bed leveling and z-axis motors are off after ~1.5mm of print

mocksoul, 2019-01-15:
Hi! New user here :)

First of all - thanks for great piece of software and (almost) great docs!

Installed on Creality3d CR10S Pro with stock capacitive nozzle leveler.

Bed leveling works nice, except 2 problems

Minor problem -- there is no way to home Z axis with bed leveling mesh turned on. Scenario:

1) home xy
2) move to the center of bed
3) home z

in 2nd step klipper complains "you need to home Z first!". So, temporarily turning off mesh, home z, turn mesh on again, home z again -- works. I dunno is it expected behaviour or no, but fighted with that for a couple of minutes.

Major problem -- in 1.5mm of print Z-axis motors are completely turned off. 

I was using 4x4 matrix, fade start 1, fade end 6, layer height 0.10.
During that  10 good layer of print I'm feeling slight Z-axis corrections during XY move. But after they are almost fade out motors just turn off and can be freely moved by hand, while print continues (and destroys extruder :))

Mesh used:

`#*# [bed_mesh main]
#*# points =
#*#     0.088333, 0.207500, 0.382500, 0.627500
#*#     -0.099167, -0.045833, 0.080833, 0.296667
#*#     -0.168333, -0.188333, -0.133333, 0.027500
#*#     -0.018333, -0.070000, -0.040833, 0.104167
#*# x_count = 4
#*# y_count = 4
#*# min_x = 50.0
#*# max_x = 269.99
#*# min_y = 30.0
#*# max_y = 270.0
#*# x_offset = 0.0
#*# y_offset = 0.0
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
`

Without mesh bed leveling everything works as expected, no problems here.

I'm not providing logs here, coz I didnt find anything there, except "Stats" prints. But if you need - I'm able to reproduce problem easily. Also tried 2us step pulse with no success. 

Didnt tried tilt bed leveling, yet.

Arksine, 2019-01-15:
Your first issue can be resolved using a [homing_override].  You can add set_position_z: 0 to it to keep it from complaining about homing z.

We will need logs for the second issue.  Nothing in mesh leveling should result in the z-axis motors turning off, if it runs upon an error it should abort the print.  I recommend disabling fade to see if that is the contributing factor.

jakep82, 2019-01-15:
Are you using a TMC2208 in standalone mode for the Z axis?

mocksoul, 2019-01-15:
> Are you using a TMC2208 in standalone mode for the Z axis?

To be honest I have no idea which drivers used in my CR10S Pro. It looks like Creality dont tell anyone, but it seems to be chinese TMC clone (A4988?)

> you are probably using too much run current resulting in the z driver overheating.

I didnt change everything related to currents. I dont even know how to do that, yet :))

> I recommend disabling fade to see if that is the contributing factor
I will try, at least several prints (12hrs+) wo bed leveling completed without any errors. I'll try disabling fade, which is kinda weird, but for test... 

> Your first issue can be resolved using a [homing_override]. You can add set_position_z: 0 to it to keep it from complaining about homing z.

I did this:

```[homing_override]
gcode: G1 F6000
       G28 X0 Y0
       BED_MESH_CLEAR
       G1 X150 Y150
       BED_MESH_PROFILE LOAD=default
       G28 Z0
axes: z```

Probably your idea to force klipper think it is at Z0 before Z homing will work too, thanks for advice!

> We will need logs for the second issue
/tmp/klipper.log is enough?

Arksine, 2019-01-15:
I typically recommend disabling fade when trying to diagnose potential issues with mesh leveling.  In your situation I do not think it will help though.

FWIW, if the default holding current is lower than the running current, moving the axes up and down or printing normally without mesh leveling may not lead to over heating.  Mesh Leveling is  is constantly adjusting the z-axis, thus the driver doesn't have a chance to drop to a holding state.  The behavior you are describing is identical to the behavior of a TMC driver when it detects a fault.  

I believe Creality is now in the habit of posting their source code, so hopefully you will be able to get the TMC settings there (I assume that is how you got your pin settings)?   Maybe the source will give you and idea of which TMC driver you are dealing with.  Trinamic drivers are very different from A4988s.  You don't configure them with potentiometers and jumpers, you configure them over serial (either UART or SPI depending on the driver).  They have default settings and that is what you are running at now.  Once you know which type of TMC driver you have (TMC2130, TMC2208, etc) you can look at examples-extra.cfg for information on how to configure them.

/tmp/klippy.log is enough, although I'm reasonably confident that your issue is due to the drivers if no error is raised.  When they are configured you can run the DUMP_TMC command to see if the driver turned off due to a fault.

jakep82, 2019-01-15:
Some quick Googling reveals that this printer does have TMC drivers, but I can't find info on what driver.  If they're 2208 drivers, that would likely explain your issue with the driver turning off in the middle of a print.

mocksoul, 2019-01-15:
Reproduced today, fade 1-10 (10 - advised in docs), also set fade_target=0 (again, advice from docs)

Failed in ~9.5mm

There is `print_stall=1` in logs, what does that mean?

```_bytes=0 freq=16000089 heater_bed: target=50 temp=50.0 pwm=0.023 print_time=2071.419 buffer_time=2.244 print_stall=0 extruder: target=205 temp=204.8 pwm=0.636
Stats 134575.2: gcodein=348158 mcu: mcu_awake=0.076 mcu_task_avg=0.000239 mcu_task_stddev=0.000232 bytes_write=1336960 bytes_read=322072 bytes_retransmit=0 bytes_invalid=0 send_seq=25054 receive_seq=25054 retransmit_seq=0 srtt=0.013 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000088 heater_bed: target=50 temp=50.0 pwm=0.096 print_time=2073.253 buffer_time=3.077 print_stall=0 extruder: target=205 temp=204.8 pwm=0.636
Stats 134576.2: gcodein=348158 mcu: mcu_awake=0.066 mcu_task_avg=0.000234 mcu_task_stddev=0.000231 bytes_write=1337005 bytes_read=322216 bytes_retransmit=0 bytes_invalid=0 send_seq=25058 receive_seq=25058 retransmit_seq=0 srtt=0.011 rttvar=0.004 rto=0.026 ready_bytes=8 stalled_bytes=0 freq=16000087 heater_bed: target=50 temp=50.0 pwm=0.178 print_time=2073.253 buffer_time=2.076 print_stall=0 extruder: target=205 temp=204.8 pwm=0.636
Stats 134577.2: gcodein=348603 mcu: mcu_awake=0.066 mcu_task_avg=0.000234 mcu_task_stddev=0.000231 bytes_write=1338736 bytes_read=322475 bytes_retransmit=0 bytes_invalid=0 send_seq=25088 receive_seq=25088 retransmit_seq=0 srtt=0.012 rttvar=0.003 rto=0.025 ready_bytes=51 stalled_bytes=0 freq=16000089 heater_bed: target=50 temp=50.0 pwm=0.034 print_time=2074.557 buffer_time=2.380 print_stall=0 extruder: target=205 temp=204.8 pwm=0.636
Stats 134578.2: gcodein=349084 mcu: mcu_awake=0.066 mcu_task_avg=0.000234 mcu_task_stddev=0.000231 bytes_write=1340680 bytes_read=322745 bytes_retransmit=0 bytes_invalid=0 send_seq=25123 receive_seq=25123 retransmit_seq=0 srtt=0.012 rttvar=0.002 rto=0.025 ready_bytes=29 stalled_bytes=0 freq=16000088 heater_bed: target=50 temp=50.0 pwm=0.063 print_time=2075.591 buffer_time=2.414 print_stall=0 extruder: target=205 temp=204.8 pwm=0.635
Stats 134579.2: gcodein=349470 mcu: mcu_awake=0.066 mcu_task_avg=0.000234 mcu_task_stddev=0.000231 bytes_write=1342355 bytes_read=323004 bytes_retransmit=0 bytes_invalid=0 send_seq=25153 receive_seq=25153 retransmit_seq=0 srtt=0.011 rttvar=0.003 rto=0.025 ready_bytes=23 stalled_bytes=0 freq=16000087 heater_bed: target=50 temp=50.0 pwm=0.117 print_time=2076.442 buffer_time=2.265 print_stall=0 extruder: target=205 temp=204.8 pwm=0.635
Stats 134580.2: gcodein=349758 mcu: mcu_awake=0.066 mcu_task_avg=0.000234 mcu_task_stddev=0.000231 bytes_write=1343634 bytes_read=323228 bytes_retransmit=0 bytes_invalid=0 send_seq=25176 receive_seq=25176 retransmit_seq=0 srtt=0.013 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000086 heater_bed: target=50 temp=50.0 pwm=0.170 print_time=2077.382 buffer_time=2.204 print_stall=0 extruder: target=205 temp=204.9 pwm=0.635
Stats 134581.2: gcodein=350281 mcu: mcu_awake=0.073 mcu_task_avg=0.000242 mcu_task_stddev=0.000234 bytes_write=1345525 bytes_read=323498 bytes_retransmit=0 bytes_invalid=0 send_seq=25208 receive_seq=25208 retransmit_seq=0 srtt=0.012 rttvar=0.003 rto=0.025 ready_bytes=23 stalled_bytes=0 freq=16000085 heater_bed: target=50 temp=49.9 pwm=0.223 print_time=2078.795 buffer_time=2.616 print_stall=0 extruder: target=205 temp=205.4 pwm=0.488
Stats 134582.2: gcodein=350656 mcu: mcu_awake=0.073 mcu_task_avg=0.000242 mcu_task_stddev=0.000234 bytes_write=1346411 bytes_read=323692 bytes_retransmit=0 bytes_invalid=0 send_seq=25225 receive_seq=25225 retransmit_seq=0 srtt=0.012 rttvar=0.003 rto=0.025 ready_bytes=46 stalled_bytes=0 freq=16000084 heater_bed: target=50 temp=50.0 pwm=0.066 print_time=2079.243 buffer_time=2.063 print_stall=0 extruder: target=205 temp=205.4 pwm=0.488
Stats 134583.2: gcodein=350878 mcu: mcu_awake=0.073 mcu_task_avg=0.000242 mcu_task_stddev=0.000234 bytes_write=1347454 bytes_read=323896 bytes_retransmit=0 bytes_invalid=0 send_seq=25244 receive_seq=25244 retransmit_seq=0 srtt=0.009 rttvar=0.004 rto=0.026 ready_bytes=0 stalled_bytes=0 freq=16000084 heater_bed: target=50 temp=50.0 pwm=0.091 print_time=2080.032 buffer_time=1.851 print_stall=0 extruder: target=205 temp=205.4 pwm=0.488
Stats 134584.2: gcodein=351021 mcu: mcu_awake=0.073 mcu_task_avg=0.000242 mcu_task_stddev=0.000234 bytes_write=1347660 bytes_read=324011 bytes_retransmit=0 bytes_invalid=0 send_seq=25248 receive_seq=25248 retransmit_seq=0 srtt=0.009 rttvar=0.005 rto=0.027 ready_bytes=0 stalled_bytes=0 freq=16000084 heater_bed: target=50 temp=50.1 pwm=0.000 print_time=2080.291 buffer_time=1.108 print_stall=0 extruder: target=205 temp=205.4 pwm=0.488
Stats 134585.2: gcodein=352093 mcu: mcu_awake=0.073 mcu_task_avg=0.000242 mcu_task_stddev=0.000234 bytes_write=1348703 bytes_read=324215 bytes_retransmit=0 bytes_invalid=0 send_seq=25267 receive_seq=25267 retransmit_seq=0 srtt=0.011 rttvar=0.002 rto=0.025 ready_bytes=8 stalled_bytes=0 freq=16000085 heater_bed: target=50 temp=50.0 pwm=0.130 print_time=2081.593 buffer_time=1.408 print_stall=1 extruder: target=205 temp=205.1 pwm=0.640
```

mocksoul, 2019-01-15:
> I believe Creality is now in the habit of posting their source code
They did. Marlin 1.1.3, and TMC drivers are not used there. Personally, I dont beleive they use TMC, since they hide this info for some reason. The only one statement - is the presentation slide https://www.creality3d.cn/u_file/1812/photo/9905c52320.jpg which does not tell which "TMC" they use =)

I tried to enable tmc 2130/2208 in marlin source code without any success several days ago.

mocksoul, 2019-01-15:
Set `fade_end=0` (i.e. disabled fade)
And xyz calibration cube (failed in prev try) printed without any problems. Z axis was constantly adjusting during print till end.

And no more suspicious "print stalls" :)

```
pi@c3d:/tmp $ grep -a print_stall= /tmp/klippy.log | awk '{print $27}' | sort | uniq -c
   2474 print_stall=0
```

(log was rotated just before test print)

offtop: last xyz cube was my most beautiful xyz cube ever printed. Klipper is a masterpiece!

mocksoul, 2019-01-15:
Cut last failed log and  put it to analyzer. Looks like host buffer suddenly jumps and "print_stall" occurs and Z-axis fail at that moment.

[image]

mocksoul, 2019-01-15:
And this is last successfull print of the same model, without fade but with bed leveling turned on (I didnt releveled bed between two tries, only fade was turned off)

[image]

mocksoul, 2019-01-15:
Both logs (good & fail)

mocksoul, 2019-01-15:
Also note: I had marlin 1.1.9 stock sources compiled and runing with mesh bed leveling and fade (0-5mm) 2 weeks with 24x7 printing without any fail on any axis (but with few skipped steps on Y and once on X -- see above). Since I didnt changed anything in marlin regarding to stepper drivers, I dont believe in "overcurrent/overheating" idea...

There was prints with huge retraction (8mm) and z-hop (2mm) and still z-axis didnt failed even once.

Arksine, 2019-01-15:
It seems likely that that the CR10s Pro is using TMC2208s in standalone mode and you are experiencing the problem described in issue #196.  Note that it doesn't affect Marlin due to the different way it generates steps.

Arksine, 2019-01-15:
Reducing accel might help.  Perhaps disabling fade to keep up/down transitions from converging may help mitigate the issue as well.

If the DIAG pin is connected to the Atmega 2560 then it should be possible to check its state to determine if there is a fault.  If there is a fault during the homing procedure then Klipper will send an error, as the axis won't retract after homing.

It would be unfortunate for Creality to design the board and not wire the 2208's UART pin to a free pin on the Atmega 2560.    I would contact Creality to see if they are willing to confirm that is the case, and if not which pin(s) are connected.

Lyr3x, 2019-01-16:
I recognised a similar problem using a MKS GEN 1.4 with TMC2130 and dual z. When the print reaches the third layer (.2 layer height) the compensation stopped (no fade activated).

mocksoul, 2019-01-17:
Reducing accel didnt help. But dramatically reducing fade distance (1..6 => 0.5..1) - "fixed" it. At least ~10 prints after that was without problem.

I did also tried pressure advance -- and same problem pops with extruder driver =). The lower advance I set - the higher chance to extruder motor driver will turn off soon. It looks like it fails if a lot of small (actually, it looks like "super small") movements scheduled.

So, 2 advance -- no problems, 1 advance - ~70-90% chance to finish without fail. 0.2 advance -- fail in first 1-5 layers (XYZ calibration cube). Of course, it depends on amount of corners in model and depends on head speed and accel. Higher head accel -- less chance to fail =)). It even sounds badly on low accel -- motor vibrates a lot without any sign of "smooth" movement. But after increasing head acceleration - pressure advance will work much smoother. Anyway I had 0 success with pressure advance, yet =(

I'll dig in a code, maybe I'll be able to decrease motor resolution in klipper. There is no reason to schedule movements motor will not be able to do.. It should be taken into account in internal calculations, but scheduled once it hits some threshold (1 step distance for extruder, for example, it is 0.007mm in my setup).

Also I have a lot of issues with retraction in bowden setup. Saw some tickets here about that. There is definitely some overextrude problem with retraction >= 5mm with >= 100mm/s speeds. Corners are fine, only retraction has issues.

I guess Marlin made optimization here -- if we have move + retraction, complete that move till stop and make retraction then is a silly thing, there will be blob at the end. Much better idea is to calculate scheduled retraction time, devide it by some factor and replace final part of movement with retraction, so head will not freeze at the end of move.

I tried crazy extruder numbers like 2000 mm/sec velocity and 5000 mm/sec^2 accel -- and stil have those freezes. 5000mm/sec accel means it should take just 45ms to travel 5mm of filament. I beleive visible freeze is much longer :). I did simple test with few bars 0.8mm width and 40mm long all placed parallel with some distance between them. It forces slicer to schedule retraction in same corner all time. FInally "non-retraction" side of bar is sharp with great corners, while the one with retraction fails completely with overextrusion. Error margin here come worse with more layers printed. Pressure advance can hide it a little bit with underextrusion just before retraction, but still I was unable to make success corners.

Marlin comletes this test with any retraction -- I tried even stupid 50mm - there is no difference in quality between 5mm and 50mm retraction. But I changed extruder between those tests (stock CR10SPro => MicroSwiss) but dont believe it play in this test big role.

This is kinda offtopic to this ticket, I'll dig deeper and maybe post ticket/pr if I'll find something usefull. Indeed everything marlin do should be possible in klipper and even more, because of better overall design :)

KevinOConnor, 2019-01-17:
The TMC2208s in standalone mode are not compatible with Klipper due to the issue described at: https://github.com/KevinOConnor/klipper/issues/196#issuecomment-416237533

There is no work around for this (other than programming them via uart).  Changing accel, changing step size, changing speed, changing pressure advance, etc., will not eliminate the issue.

-Kevin

mocksoul, 2019-01-18:
closed, since it seems to be complete duplicate of #196

closed 2019-01-18 by mocksoul