Klipper3d/klipper #1114: Display no Menu on screen - fysetc f6 v1.3 with tmc2208 v1.2

bruce356, 2019-01-15:
Hi, just running the Generic fysetc F6 configuration printer.cfg (thanks to Simone Bonacini) and default menu.cfg . Not sure what the problem might be with my setup but I have a "RepRapDiscount 128x64 Full Graphic Smart Controller", menus do not appear, just a blank screen with background light on.

With the same cabling, board etc. the display works as expected with Marlin.
Hoping someone can shed some light on this - thanks Bruce

bruce356, 2019-01-17:
Hi for anyone else who might have the same display with FYSETC F6 v1.3, this worked for me:-

# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
[display]
lcd_type: st7920	;12864
cs_pin: PH1
sclk_pin: PA1
sid_pin: PH0
menu_timeout: 60
menu_root: __main	
encoder_pins: ^PC6, ^PC4
click_pin: ^!PC2
kill_pin: ^!PG0

Bruce

KevinOConnor, 2019-01-17:
@SimoneBnc - do you know why there would be a difference in st7920 pin mapping?

-Kevin

SimoneBnc, 2019-01-17:
That's interesting. I used the display that came on my ender 3 and flipped 180° the lcd cable connector on the control board side as seen on many guides, so unless there are some differences on the 2 display we are talking about I don't know. I used the pins configuration from the other issue you linked to me the other day. I may give this new pin config a try when I get home

bruce356, 2019-01-19:
I am sure my Mapping is correct
[image]
Regards - Bruce

LazaroFilm, 2019-01-19:
Some boards have the connectors mounted backwards. You need to pull the plastic cover off and slip it [as explained here.](https://youtu.be/ZvRfaYAIc4g)

bruce356, 2019-01-20:
Hi thanks but I am aware of the connector issue, this was not a connector issue but the wrong pin mapping for the  RepRapDiscount 128x64 Full Graphic Smart Controller 
regards - bruce

closed 2019-01-21 by bruce356

KevinOConnor, 2019-01-25:
So, was this resolved?

-Kevin

m24andrew, 2019-02-09:
There must be a difference between the displays.  The version in the config by Simone Bonacini works correctly for the display that comes with the Ender 3 attached to the Fysetc F6 board.  With the new pin configuration from #1206 the Ender 3 stops working.

nandinho84, 2019-02-11:
Yes maybe... Mine was bought from an aliexpress supplyer and works well with the config from #1206 .
Maybe beacause your's for a specific 3d printer, maybe the pin mapping is different. Once I had a display for the Anet A6 and the pin mapping was different from the generic reprap full graphics display, maybe this is your case... Use the config that suits you most

SimoneBnc, 2019-02-11:
There is no real standard for this kind of no brand lcd displays, so anything could be wrong or correct depending on the specific case

KevinOConnor, 2019-02-14:
Well, that's very surprising. The fysetc wiki indicates the pins match RAMPS, which is what #1206 implements.  It's not clear if some fysetc boards are different or if some LCD displays are different. It would be great if someone could chase this down.

-Kevin

nandinho84, 2019-02-14:
I have two fysetc F6 boards and they work ok with the #1206 pin assignment. As I told before, a few months ago I had a LCD 12864 from the Anet A6... To use it with the ramps I needed to make some pin changes, because it didn't match the usual 12864 LCD's (

trevjonez, 2019-07-09:
I picked up a few of these  full graphic smart controllers on amazon and the connectors had to be turned over to get it to work. IE the guide tab is on the bottom and needs to be towards the top