Klipper3d/klipper #1115: Ziegler-Nichols autotune pid implementation question

pmarkiewicz, 2019-01-15:
I have question about Ziegler-Nichols implementation in Klipper.
Looking at https://en.wikipedia.org/wiki/Ziegler%E2%80%93Nichols_method Ti and Td depends only on Tu.
But in klipper https://github.com/KevinOConnor/klipper/blob/7fac0de1f879b6a304bdb3f8f3acfdab6145608d/klippy/extras/pid_calibrate.py Ti an Td also depends on Kp.

Is there any better explanation why this implementation is different than present in wikipedia and some other sources?

KevinOConnor, 2019-01-15:
It's still common to use the Ziegler–Nichols formulas for PID parameters (their original work is from 1942).  However, it's quite rare to use their original tuning method to find Ku and Tu as using it is difficult and potentially dangerous.  Instead, it's now common to use the "Astrom-Hagglund relay method" to estimate Ku and Tu (this was published in 1984).  Once Ku and Tu are estimated, they can be used with the original Ziegler–Nichols formulas.

Not sure what you mean by "Ti and Td depend on Kp" - Klipper does not directly do that.

-Kevin

KevinOConnor, 2019-01-16:
If you look at the wikipedia article for Ziegler–Nichols ( https://en.wikipedia.org/wiki/Ziegler%E2%80%93Nichols_method ) note that it describes the output of the control system as:
`heater_output = Kp * ( error + (1/Ti)*integral_error + Td*derivative_error )`

Klipper (in keeping with other popular firmware) instead uses this form of the pid:
`heater_output = Kp*error + Ki*integral_error + Kd*derivative_error`
Thus, Ki and Kd are calculated in PID_CALIBRATE accordingly.

-Kevin