Klipper3d/klipper #1117: Trying to configure a CoreXY, Dual Z, TMC2130 on MKS Gen 1.4

Lyr3x, 2019-01-15:
I have some problems to port my Marlin Configuration to a working Klipper config. My Setup is as follows:
- Core XY
- Dual Z (second Stepper is connected to E1)
- Five TMC2130 
Problems:
- corexy kinematics are not enabled (Actually X and Y moving in the same direction
- Only one Z stepper is working
Here is the relevant part:
```
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: tmc2130_stepper_x:virtual_endstop
#endstop_pin: ^ar2
position_endstop: 0
position_max: 300
homing_speed: 50

[tmc2130 stepper_x]
cs_pin: ar59
microsteps: 16
run_current: 0.950
hold_current: 0.500
stealthchop_threshold: 100
driver_SGT: 3
diag1_pin: ^!ar3
sense_resistor: 0.11

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: tmc2130_stepper_y:virtual_endstop
#endstop_pin: ^ar15
position_endstop: 0
position_max: 300
homing_speed: 50

[tmc2130 stepper_y]
cs_pin: ar63
microsteps: 32
run_current: 0.950
hold_current: 0.500
stealthchop_threshold: 200
driver_SGT: 5
diag1_pin: ^!ar14
sense_resistor: 0.11

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
#endstop_pin: ^ar19
position_max: 270
position_min: -2

[tmc2130 stepper_z]
cs_pin: ar40
microsteps: 16
run_current: .800
sense_resistor: 0.11

[stepper_z1]
step_pin: ar36
dir_pin: ar34
enable_pin: ar30
step_distance: .0025
endstop_pin: ^ar19

[tmc2130 stepper_z1]
cs_pin: ar44
microsteps: 16
run_current: .8
sense_resistor: 0.11

[printer]
kinematics: corexy
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100
```

jakep82, 2019-01-15:
You need to define your endstop pin for stepper_z (^ar19), and comment it out for stepper_z1.  You aren't using a probe, so you don't have a virtual endstop.  Both Z motors will stop when that pin is triggered.

For the XY steppers, just change the dir_pin on one of them if it's running the wrong direction.  Adding a removing a ! will change the direction.

Lyr3x, 2019-01-15:
I just let the probe section out here. I configured a bltouch

jakep82, 2019-01-15:
So are you using an endstop switch, or are you using the probe as an endstop?  The way you have it set up, you're using the bltouch as an endstop for stepper_z, and a separate endstop switch for stepper_z1. That's not a proper configuration.  Also, have you run query_endstops?  

I would suggest running through [this](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md) document if it's a new printer to confirm everything is configured correctly.

Lyr3x, 2019-01-15:
The Z1 endstop is a misconfiguration. I'll check the document 👍

Lyr3x, 2019-01-15:
Alright i changed the direction of the x axis but this did not solve the issue regarding the corexy kinematics. Testing with
STEPPER_BUZZ STEPPER=stepper_x: results now in an diagonal move from mid to 0/0
STEPPER_BUZZ STEPPER=stepper_y: only the B motor turns

changed z stepper to:
```
[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
#endstop_pin: ^ar19
position_max: 270
position_min: -2

[tmc2130 stepper_z]
cs_pin: ar40
microsteps: 16
run_current: .800
sense_resistor: 0.11

[stepper_z1]
step_pin: ar36
dir_pin: ar34
enable_pin: ar30
step_distance: .0025

[tmc2130 stepper_z1]
cs_pin: ar44
microsteps: 16
run_current: .8
sense_resistor: 0.11
```

But still no double z movement

jakep82, 2019-01-15:
That is correct regarding stepper_buzz.  Kinematics are not loaded, so it only move the stepper you've define as X or Y.  I diagonal move is expected.  Have you tried running G28 X0 to see if it moves correctly?

Regarding stepper_z1, I think you need a ! on your enable pin.

jakep82, 2019-01-15:
I would also strongly recommend using the same stealtchop threshold for X and Y.

Lyr3x, 2019-01-15:
Damn, i am blind.. Sorry dude 👎 Already changed the stealthchop values. X/Y moving fine now but the second z motor doesnt do anything

jakep82, 2019-01-15:
Did you change the enable pin on stepper_z1?  Are you sure all the pins for both Z steppers are correct?  Which one doesn't move?

Lyr3x, 2019-01-15:
So here are the updated config part:
````
[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
#endstop_pin: ^ar19
position_max: 270
position_min: -2

[tmc2130 stepper_z]
cs_pin: ar40
microsteps: 16
run_current: .800
sense_resistor: 0.11

[stepper_z1]
step_pin: ar36
dir_pin: ar34
enable_pin: !ar30
step_distance: .0025

[tmc2130 stepper_z1]
cs_pin: ar44
microsteps: 16
run_current: .800
sense_resistor: 0.11
```

RAMPS.h:

```
#define E1_STEP_PIN        36
#define E1_DIR_PIN         34
#define E1_ENABLE_PIN      30
#ifndef E1_CS_PIN
  #define E1_CS_PIN        44
#endif
```

jakep82, 2019-01-15:
Are you using a jumper wire for the CS pin?  Are you sure it's connected to ar44?  You could also try switching motor connections on the board to verify that the motor works.

Lyr3x, 2019-01-15:
Weird, i reflashed the MCU part and now the Z Motors are working. Not sure what was wrong there. 

I just need to get the BLTouch working and im good to go. Servo is working fine but the endstop is not yet

jakep82, 2019-01-15:
If it's a clone, good luck.  You'll find a length issue here regarding clones, and some are not working well.  If it's genuine, I would delete the probe and servo sections in your config, and only use the bltouch section.  It's easier to configure and works well with a genuine bltouch

Lyr3x, 2019-01-15:
Okay i'll change that because its a genuine. Thank you!

closed 2019-01-15 by Lyr3x