Klipper3d/klipper #1118: Wierd X and Y problems while no error message

CharlesStArnault, 2019-01-15:
So I was tweaking with Pid settings on my hotend and when done, (please note I didn't change anything for the X and Y axis) I saved and when using my printer it seemed the x and y shifted. When I try to home, the x move on y axis and y on x axis. When I try to change the dir pins, it just does the same thing but in the opposite direction. Now where it gets interesting: when I press the limit switches manually so octoprint thinks it is on home position, when I try to move the hotend with the arrows, I Get weird results. The right arrow moves the hotend to the right, which is normal. The top arrow moves the hotend to the top, which again is normal. But the left arrow also moves the hot end up. And the bottom arrow moves the hotend to the right. I have a mks 1.4 board with core XY printer. I have attached my config file. I omitted the log file because I am not having any issue with the actual software or anything related, so it isn't useful, at least in my opinion. 

[config.zip](https://github.com/KevinOConnor/klipper/files/2761562/config.zip)

jakep82, 2019-01-15:
Everything looks okay in the config.  It would be better to type commands in the terminal so you know exactly what is being sent to the printer.  It's possible you've inadvertently changed something in Octoprint which is causing an issue with the control buttons.

KevinOConnor, 2019-01-21:
There's little we can do to assist without the full klipper log file - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Beyond that, the only generic advice I can give is to rerun the configuration checks described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

closed 2019-02-03 by KevinOConnor